UniversalRobots/Universal_Robots_ROS2_Driver #1294: `YAML_CPP_LIBRARIES` is empty in `ur_calibration` causing the `yaml-cpp` library  to be missing while during linking

razr, 2025-03-13:
### Affected ROS2 Driver version(s)

humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Other

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

Real robot

### Robot SW / URSim version(s)

n/a

### How is the ROS driver used.

Others

### Issue details

# Summary
`YAML_CPP_LIBRARIES` is empty in `ur_calibration` causing the `yaml-cpp` library  to be missing while during linking

## Issue details
I build `humble` from source and use [`yaml_cpp_vendor`](https://github.com/ros2/yaml_cpp_vendor/tree/humble) to produce `yaml-cpp`. And `YAML_CPP_LIBRARIES` is empty.

```bash
$ root/share/cmake/yaml-cpp/yaml-cpp-config.cmake
# - Config file for the yaml-cpp package
# It defines the following variables
#  YAML_CPP_INCLUDE_DIR - include directory
#  YAML_CPP_LIBRARIES    - libraries to link against

# Compute paths
get_filename_component(YAML_CPP_CMAKE_DIR "${CMAKE_CURRENT_LIST_FILE}" PATH)
set(YAML_CPP_INCLUDE_DIR "")

# Our library dependencies (contains definitions for IMPORTED targets)
include("${YAML_CPP_CMAKE_DIR}/yaml-cpp-targets.cmake")

# These are IMPORTED targets created by yaml-cpp-targets.cmake
set(YAML_CPP_LIBRARIES "")
```

`yaml-cpp` is defined as an imported target with `IMPORTED_LOCATION_RELEASE` and `CMake` expects a project to link against it using:

```bash
target_link_libraries(${PROJECT_NAME} yaml-cpp)
```

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior

`yaml-cpp` is defined as an imported target with `IMPORTED_LOCATION_RELEASE` and `CMake` expects a project to link against it using:

```bash
target_link_libraries(${PROJECT_NAME} yaml-cpp)
```

Even `yaml-cpp::yaml-cpp` as recommended [here](https://github.com/ros2/yaml_cpp_vendor/issues/51#issuecomment-2149675576)

## Actual Behavior

see above

# Workaround Suggestion

see above

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-03-13:
Thank you for pointing that out. That seems to be an old tech debt to update.

razr, 2025-03-13:
Thanks @urfeex I'll test it later today.

urfeex, 2025-03-14:
Let's go with that one then, thanks for testing.

PR #1304 (2025-03-18): Use modern CMake to link against yaml-cpp (backport #1295)
The way we link against yaml-cpp seems to be a leftover from times where it didn't support modern CMake.

Fixes #1294 <hr>This is an automatic backport of pull request #1295 done by [Mergify](https://mergify.com).
closed 2025-03-18 by urfeex