UniversalRobots/Universal_Robots_ROS2_Driver #1296: Connection to UR5e using a VM fail

marols27, 2025-03-13:
### Affected ROS2 Driver version(s)

3.0.2 and 3.1.0

### Used ROS distribution.

Other

### Which combination of platform is the ROS driver running on.

Linux in a virtual machine

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.12

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
*We are attempting to connect the ROS2 driver to a ur5e robot from a virtual ubuntu machine for testing. When we have the driver and connection established, we will also connect the MiR by ROS2 as well and then we will attempt to create a collaboration setup between the two robots. We are using a virtual machine for better system version handling, for easier rollbacks if something does not work, and because we up till now did not have any other options. 

Noteworthy: We have been able to connect UR5e using the driver, however, only to read its position, not to control it. *

## Issue details
We are now running an Ubuntu 24.02, with an instance of Jazzy Jalisco using VMWare 16 Player on Windows.
We have followed each of the setup steps at the driver home page: [https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver](url).
And when we attempt a connection by the prompt "ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:= 172.31.1.144" in our VM, we get the following output:

## Steps to Reproduce
- Create an Ubuntu 24.02 VM in VMWare.
- Install the Jazzy Jalisco ros2.
- Do the setup as instructed at the home page [https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver](url).
- Runn the launch file.

## Expected Behavior
*We expected it to connect to UR5e and gain ability to run controller programs and the URCaps program to connect in return. *

## Actual Behavior
*It does not connect, at least not reliably, and we are unable to control the robot and run the URCaps control script. *

# Workaround Suggestion
*We are aware that the VM part might be the issue (we have been informed), and to solve it we would have to get a computer with Ubuntu installed as the host system and run ROS2 directly on the host. 

However, the ability to run the driver would be greatly helpful as we would easier be able to roll back systems and updates in case of accidents. *

### Relevant log output

```shell
[INFO] [launch]: All log files can be found below /home/prosjekt/.ros/log/2025-03-12-15-22-10-883220-ROS2-jazzy-jalisco-318102
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [318120]
[INFO] [controller_stopper_node-3]: process started with pid [318121]
[INFO] [ros2_control_node-1]: process started with pid [318119]
[INFO] [urscript_interface-4]: process started with pid [318122]
[INFO] [robot_state_publisher-5]: process started with pid [318123]
[INFO] [rviz2-6]: process started with pid [318124]
[INFO] [spawner-7]: process started with pid [318125]
[INFO] [spawner-8]: process started with pid [318126]
[controller_stopper_node-3] [INFO] [1741789331.557913141] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[robot_state_publisher-5] [INFO] [1741789331.775876782] [robot_state_publisher]: Robot initialized
[dashboard_client-2] [INFO] [1741789331.870086355] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[ros2_control_node-1] [INFO] [1741789331.915801683] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[ros2_control_node-1] [INFO] [1741789331.917017232] [controller_manager]: update rate is 500 Hz
[ros2_control_node-1] [INFO] [1741789331.917029175] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-1] [WARN] [1741789331.917359164] [controller_manager]: Could not enable FIFO RT scheduling policy: with error number <1>(Operation not permitted). See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[ros2_control_node-1] [INFO] [1741789331.934038308] [controller_manager]: Received robot description from topic.
[ros2_control_node-1] text not specified in the tf_prefix tag
[ros2_control_node-1] [INFO] [1741789331.950969098] [controller_manager.resource_manager]: Loading hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1741789331.968873254] [controller_manager.resource_manager]: Loaded hardware 'ur5e' from plugin 'ur_robot_driver/URPositionHardwareInterface'
[ros2_control_node-1] [INFO] [1741789331.968998070] [controller_manager.resource_manager]: Initialize hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1741789331.969182298] [controller_manager.resource_manager]: Successful initialization of hardware 'ur5e'
[ros2_control_node-1] [INFO] [1741789331.974429857] [resource_manager]: 'configure' hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1741789331.974511362] [URPositionHardwareInterface]: Starting ...please wait...
[ros2_control_node-1] [INFO] [1741789331.974531660] [URPositionHardwareInterface]: Initializing driver...
[ros2_control_node-1] [WARN] [1741789332.020568076] [UR_Client_Library:]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_client_library/doc/real_time.html for details.
[ros2_control_node-1] [INFO] [1741789332.277800543] [UR_Client_Library:]: Negotiated RTDE protocol version to 2.
[ros2_control_node-1] [INFO] [1741789332.278663395] [UR_Client_Library:]: Setting up RTDE communication with frequency 500.000000
[rviz2-6] [INFO] [1741789332.667587987] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-6] [INFO] [1741789332.667795690] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[rviz2-6] [INFO] [1741789332.837005320] [rviz2]: Stereo is NOT SUPPORTED
[ros2_control_node-1] [INFO] [1741789333.322072015] [URPositionHardwareInterface]: Calibration checksum: 'calib_12788084448423163542'.
[spawner-8] [INFO] [1741789333.595712427] [spawner_joint_trajectory_controller]: waiting for service /controller_manager/list_controllers to become available...
[spawner-7] [INFO] [1741789333.631271237] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[ros2_control_node-1] [ERROR] [1741789334.337255646] [URPositionHardwareInterface]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_calibration/README.md] for details.
[ros2_control_node-1] [INFO] [1741789334.337354553] [URPositionHardwareInterface]: System successfully started!
[ERROR] [ros2_control_node-1]: process has died [pid 318119, exit code -11, cmd '/opt/ros/jazzy/lib/controller_manager/ros2_control_node --ros-args --params-file /home/prosjekt/ros2_ws/install/ur_robot_driver/share/ur_robot_driver/config/ur5e_update_rate.yaml --params-file /tmp/launch_params_ya4pkx24'].
[spawner-8] [FATAL] [1741789343.689063474] [spawner_joint_trajectory_controller]: Could not contact service /controller_manager/list_controllers
[spawner-7] [FATAL] [1741789343.707990514] [spawner_joint_state_broadcaster]: Could not contact service /controller_manager/list_controllers
[ERROR] [spawner-8]: process has died [pid 318126, exit code 1, cmd '/opt/ros/jazzy/lib/controller_manager/spawner --controller-manager /controller_manager --controller-manager-timeout 10 --inactive joint_trajectory_controller forward_velocity_controller forward_position_controller force_mode_controller passthrough_trajectory_controller freedrive_mode_controller --ros-args'].
[ERROR] [spawner-7]: process has died [pid 318125, exit code 1, cmd '/opt/ros/jazzy/lib/controller_manager/spawner --controller-manager /controller_manager --controller-manager-timeout 10 joint_state_broadcaster io_and_status_controller speed_scaling_state_broadcaster force_torque_sensor_broadcaster tcp_pose_broadcaster ur_configuration_controller scaled_joint_trajectory_controller --ros-args'].
```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-03-13:
As you use the teach pendant to start the robot: At what time does the driver die? When you press "play" on the teach pendant or before?

Since you are running in a VM (which we don't recommend for stability reasons), the VM's network configuration will be highly relevant for getting the driver to run. The fact that you are able to read the robot's joint states but cannot control it relates to the fact that for controlling the robot, it has to connect to the ROS machine. It will try to do so using the IP address through which the ROS pc makes contact with the robot. That is probably the host's IP address, which is not necessarily the same as the guest's IP address. You'll have to make sure that your guest is reachable from the outside world and pass that IP address as [`reverse_ip`](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/57d8fab880f1df6634c5cb7b5e9cbba0581d6db0/ur_robot_driver/launch/ur_control.launch.py#L445) to the driver. Any firewall that might come in between has to make sure to open ports 50001-50004 to make things work.

However, all of that doesn't necessarily explain the crash you are experiencing. Hence my initial question: Did you try to start the program on the TP and then the driver crashed or is that independent?

marols27, 2025-03-13:
We are using bridged mode to a physical network card on the host to connect to UR5e. I was not aware, so the physical adapter was configured with a working IP-address but the gest was not. I was able to ping from the guest but not connect. I set a different IP on the guest and tried and got a closer result. I am now connected with the robot and I can attempt the External Control program with URCaps, but the familiar port problem, how do I port forward and pass the IP as reverse_IP?

The ros2 system is now continuesly giving me '[ros2_control_node-1] [WARN] [1741870492.786192891] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization'. 
There are also these errors appearing in the console between the above [WARN] lines:
'
[spawner-9] [FATAL] [1741870440.128460299] [spawner_joint_trajectory_controller]: Could not contact service /controller_manager/list_controllers
[spawner-8] [FATAL] [1741870440.146133408] [spawner_joint_state_broadcaster]: Could not contact service /controller_manager/list_controllers
[ERROR] [spawner-8]: process has died [pid 19800, exit code 1, cmd '/opt/ros/jazzy/lib/controller_manager/spawner --controller-manager /controller_manager --controller-manager-timeout 10 joint_state_broadcaster io_and_status_controller speed_scaling_state_broadcaster force_torque_sensor_broadcaster tcp_pose_broadcaster ur_configuration_controller scaled_joint_trajectory_controller --ros-args'].
[ERROR] [spawner-9]: process has died [pid 19801, exit code 1, cmd '/opt/ros/jazzy/lib/controller_manager/spawner --controller-manager /controller_manager --controller-manager-timeout 10 --inactive joint_trajectory_controller forward_velocity_controller forward_position_controller force_mode_controller passthrough_trajectory_controller freedrive_mode_controller --ros-args'].
'

urfeex, 2025-03-13:
The spawner errors are a result from the control node (which contains the controller_manager) dying.

I cannot really assist in setting up a proper network setup with the VM. If you use a bridged network, your VM shows up in the network as an own participant. If that is the case, you probably won't have to setup the reverse ip.

>  set a different IP on the guest and tried and got a closer result. I am now connected with the robot and I can attempt the External Control program with URCaps

I am unsure whether I understand that. Is the startup different to before? Please always provide full log outputs, also if only a couple of lines are different.

marols27, 2025-03-13:
I'm not sure why the process would be dying, is there any way we can prevent it?

> > set a different IP on the guest and tried and got a closer result. I am now connected with the robot and I can attempt the External Control program with URCaps

> I am unsure whether I understand that. Is the startup different to before? Please always provide full log outputs, also if only a couple of lines are different.

The bridge seam to work when I assign IP-addresses to both host and guest, I did not do that before.
Now I think I am sort of connected with the driver, but here is the complete console output at this point (sorry for the incomplete one earlier):

'[INFO] [launch]: All log files can be found below /home/prosjekt/.ros/log/2025-03-13-14-28-17-007759-ROS2-jazzy-jalisco-20252
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [20257]
[INFO] [controller_stopper_node-4]: process started with pid [20259]
[INFO] [ros2_control_node-1]: process started with pid [20256]
[INFO] [robot_state_helper-3]: process started with pid [20258]
[INFO] [urscript_interface-5]: process started with pid [20260]
[INFO] [robot_state_publisher-6]: process started with pid [20261]
[INFO] [rviz2-7]: process started with pid [20262]
[INFO] [spawner-8]: process started with pid [20263]
[INFO] [spawner-9]: process started with pid [20264]
[controller_stopper_node-4] [INFO] [1741872498.148838613] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[robot_state_publisher-6] [INFO] [1741872498.481744903] [robot_state_publisher]: Robot initialized
[ros2_control_node-1] [INFO] [1741872498.526489244] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[ros2_control_node-1] [INFO] [1741872498.526871630] [controller_manager]: update rate is 500 Hz
[ros2_control_node-1] [INFO] [1741872498.527880870] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-1] [WARN] [1741872498.531541865] [controller_manager]: Could not enable FIFO RT scheduling policy: with error number <1>(Operation not permitted). See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[ros2_control_node-1] [INFO] [1741872498.546391480] [controller_manager]: Received robot description from topic.
[ros2_control_node-1] text not specified in the tf_prefix tag
[dashboard_client-2] [INFO] [1741872498.584901431] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[ros2_control_node-1] [INFO] [1741872498.617449571] [controller_manager.resource_manager]: Loading hardware 'ur5e' 
[ros2_control_node-1] [ERROR] [1741872498.634172223] [controller_manager.resource_manager]: Caught exception of type : N9pluginlib20LibraryLoadExceptionE while loading hardware: Failed to load library /opt/ros/jazzy/lib/libur_robot_driver_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: libpal_statistics.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99
[ros2_control_node-1] [WARN] [1741872498.634320290] [controller_manager]: Could not load and initialize hardware. Please check previous output for more details. After you have corrected your URDF, try to publish robot description again.
[ros2_control_node-1] [WARN] [1741872499.522100548] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[spawner-8] [INFO] [1741872499.818672075] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[rviz2-7] [INFO] [1741872499.901994500] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-7] [INFO] [1741872499.902260238] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[rviz2-7] [INFO] [1741872500.133709991] [rviz2]: Stereo is NOT SUPPORTED
[spawner-9] [INFO] [1741872500.135593679] [spawner_joint_trajectory_controller]: waiting for service /controller_manager/list_controllers to become available...
[ros2_control_node-1] [WARN] [1741872500.521305906] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1741872501.521309598] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1741872502.520784024] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1741872503.529702355] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1741872504.523314913] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1741872505.521924091] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1741872506.520968293] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1741872507.520965904] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[... cut ...]
[ros2_control_node-1] [WARN] [1741872511.521486278] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1741872512.520874832] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1741872513.520993277] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1741872514.521190503] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization'

urfeex, 2025-03-13:
```
[ros2_control_node-1] [ERROR] [1741872498.634172223] [controller_manager.resource_manager]: Caught exception of type : N9pluginlib20LibraryLoadExceptionE while loading hardware: Failed to load library /opt/ros/jazzy/lib/libur_robot_driver_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: libpal_statistics.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99
```

That looks like there is something wrong with sourcing your workspace. Is your system completely up to date? This error didn't show up earlier, right?

urfeex, 2025-03-24:
@marols27 Any updates on this?

marols27, 2025-03-24:
No, sorry i have taken a break from it, we are working on multiple parts of the project all at once.

Sendt fra Outlook for iOS<https://aka.ms/o0ukef>