UniversalRobots/Universal_Robots_ROS2_Driver #1305: Trying to control UR10e in URSim using ROS2 commands/interface

RobJMal, 2025-03-18:
Hello!

I'm trying to interface with a simulated UR10e using the URSim because I want to do some experiments in simulation before I test it on the physical robot. I'm able to setup the Docker container and have it run and interface it in the VNC, but I'm not able to have the `ur_robot_driver` examples interface and control it. I am able to control the robot arm using the simulated Teach Pendant and have it displayed in Rviz, but when I try to send commands from any ROS2 node to the simulated UR10e, it fails. How can I get this to work? 

[image]

This is the command that I ran to launch the URSim:
```
ros2 run ur_client_library start_ursim.sh -m ur10e -i 192.168.137.200
```

This is the example code that I want to run from the `ur_robot_driver` to command the robot: 
```
ros2 launch ur_robot_driver test_scaled_joint_trajectory_controller.launch.py
```

I was following the instructions here 
- https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_client_library/doc/setup/robot_setup.html#install-urcap

My setup is the following
- Ubuntu 22.04
- ROS2 Humble 
- All of these are running locally. There is a discovery server in my computer that is running with IP address `192.168.137.100`. I have verified that this IP won't collide with any local networks. 

Thank you and please let me know if there is any more information that I can provide for clarity.

urfeex, 2025-03-19:
Hi, from your description I get that

- You can start a URSim robot and there is the external control URCap on that
- You can start the driver using `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.137.200`
- The robot shows up in RViz correctly moving if you move the robot using the (virtual) teach pendant
- Sending commands to the robot "fails" -- what exactly do you mean by that? Please always provide log output so that we can see what is going wrong.

What I assume is that
 
- You are missing the external control program running on the robot
  - Have you setup a program as explained [here](https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_client_library/doc/setup/robot_setup.html#install-urcap)?
  - Did you setup the IP address of your ROS PC in the URCap settings as explained [here](https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_client_library/doc/setup/robot_setup.html#install-urcap)?
  - Did you run that program as explained [here](https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/usage/startup.html#finish-startup-on-the-robot)?

RobJMal, 2025-03-19:
@urfeex thanks for your response!

What I mean by "sending commands to the robot fails" is that when I publish a command on a ROS topic to the robot, it doesn't move. 

I did try to follow the instructions for setting the external control program to run on the robot, but the `External Control` URCap never seems to be active. Even if I remove the external URCap and restart as it suggests, the URSim doesn't make itself restart. And everytime I relaunch the URSim docker container, the URCap external controller is always adding itself back. What makes me confused even more is that because this is in simulation how can I work with this. Do you have any suggestions?

RobJMal, 2025-03-19:
After digging around, another approach I found that works and could better fit my use case (since I want to primarily control and interact with the UR10e via ROS2) is I'm running the ROS2 driver mode in `headless_mode` and the Robot's mode in `remote_control_mode`. 

In the URSim Docker container, when I use the VNC, I switch the Teach Pendant to remote control mode (refer to image below):
[image]

In another terminal, I run the ur robot driver: 
`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10e robot_ip:=192.168.137.200 headless_mode:=true`. This launches the driver and I'm able to see an RViz model of the robot whose pose matches that in the URSim. 

I then run the command `ros2 service call /io_and_status_controller/resend_robot_program std_srvs/srv/Trigger {}` and I see the following logs: 
```
[ur_ros2_control_node-1] [INFO] [1742375361.282695122] [UR_Client_Library:]: Connection to reverse interface dropped.
[ur_ros2_control_node-1] [INFO] [1742375361.285410900] [controller_manager]: Switch controller timeout is set to 0, using default 1s!
[ur_ros2_control_node-1] [INFO] [1742375361.299337947] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.
[ur_ros2_control_node-1] [INFO] [1742375361.317285794] [io_and_status_controller]: Successfully resent robot program
```
Afterwards, I try to send some example joint commands from the tutorial ([click here](https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/usage/move.html)) and I begin to see the robot move. 

### Problem with using another controller
Ultimately though, what I'm trying to do is use this [cartesian controller](https://github.com/stefanscherzinger/cartesian_controllers_universal_robots/tree/ros2) to control a simulated version of the UR10e. Since I'm building upon what I did in the previous steps, I modified the `setup.urdf.xacro` to have a field `headless_mode="true"`

```
<?xml version="1.0"?>

<robot name="setup" xmlns:xacro="http://ros.org/wiki/xacro">

        <xacro:include filename="$(find ur_description)/urdf/ur_macro.xacro" />
        <xacro:arg name="robot_ip" default="0.0.0.0" />

        <link name="world" />

        <joint name="world_joint" type="fixed">
                <origin xyz="0 0 0" rpy="0 0 0" />
                <parent link="world" />
                <child link="base_link" />
        </joint>

        <xacro:ur_robot name="ur10e"
                tf_prefix=""
                parent="world"
                robot_ip="$(arg robot_ip)"
                joint_limits_parameters_file="$(find ur_description)/config/ur10e/joint_limits.yaml"
                kinematics_parameters_file="$(find cartesian_controllers_universal_robots)/config/test_calibration.yaml"
                physical_parameters_file="$(find ur_description)/config/ur10e/physical_parameters.yaml"
                visual_parameters_file="$(find ur_description)/config/ur10e/visual_parameters.yaml"
                script_filename="$(find ur_client_library)/resources/external_control.urscript"
                input_recipe_filename="$(find ur_robot_driver)/resources/rtde_input_recipe.txt"
                output_recipe_filename="$(find ur_robot_driver)/resources/rtde_output_recipe.txt"
                headless_mode="true"
                use_fake_hardware="false">
          <origin xyz="0 0.0 0.0" rpy="0 0 0" />
        </xacro:ur_robot>

</robot>
```

I launch the file using the command `ros2 launch cartesian_controllers_universal_robots robot.launch.py robot_ip:=192.168.137.200`. However, in the terminal where I launch it, I see the following logs:
```
[spawner-5] [INFO] [1742376323.078435977] [spawner_force_torque_sensor_broadcaster]: Configured and activated force_torque_sensor_broadcaster
[INFO] [spawner-8]: process has finished cleanly [pid 35855]
[spawner-4] [INFO] [1742376323.098426734] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[spawner-6] [INFO] [1742376323.119930736] [spawner_scaled_joint_trajectory_controller]: Configured and activated scaled_joint_trajectory_controller
[INFO] [spawner-10]: process has finished cleanly [pid 35859]
[INFO] [spawner-9]: process has finished cleanly [pid 35857]
[INFO] [spawner-4]: process has finished cleanly [pid 35847]
[INFO] [spawner-5]: process has finished cleanly [pid 35849]
[INFO] [spawner-6]: process has finished cleanly [pid 35851]
[ur_ros2_control_node-2] [INFO] [1742376324.171204203] [UR_Client_Library:]: Connection to reverse interface dropped.
```

Going to the URSim VNC, this is what I see: 
[image]

When I click "Stop Program" or "Continue", I see the following:
[image]

I'm not able to see the service `/io_and_status_controller/resend_robot_program` when I do `ros2 service list`, and therefore not able to send the trigger to resend the robot program (using the command `ros2 service call /io_and_status_controller/resend_robot_program std_srvs/srv/Trigger {}`). 

Would you have any thoughts on this? I'm also sharing this on the original repo, but I thought I'd put it here if anyone has any thoughts.

urfeex, 2025-03-19:
As the cartesian_controllers are not part of this package we cannot really give support for that. The popup that you see means that the controller is sending too large motion commands to the robot which are not physically possible to execute. Please ask its authors about that.

As you got the driver running correctly, I'll close this.

However, let me add one note: From what I can see, you did not create and start a **program** containing the external control node. The URCap is installed, yes. That happens automatically in the URSim startup script.

When not using headless mode you'll have to setup the URCap in the installation as described [here](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/_images/es_07_installation_excontrol.png), create a program as depictured [here](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/_images/es_10_prog_structure_urcaps.png) and [here](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/_images/es_11_program_view_excontrol.png) and then actually start that program.

closed 2025-03-19 by urfeex