UniversalRobots/Universal_Robots_ROS2_Driver #1308: Incomplete [INFO] returned by [UR_Client_Library] when external control is started

leoCUHKSZ, 2025-03-21:
I tried to control UR5 using moveit2, everything was done according to the official documentation, but I couldn't solve the problem:

[ur_ros2_control_node-1] [INFO] [1742545003.805560723] [UR_Client_Library:]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1742545003.805628436] [UR_Client_Library:]: Sent program to robot
[ur_ros2_control_node-1] [INFO] [1742545014.027802082] [UR_Client_Library:]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1742545014.027862127] [UR_Client_Library:]: Sent program to robot

When I click to start external control, the information returned is still incomplete,  as far as I know it should return a third message indicating that the robot arm is ready to receive the control command.

The full log is as follow:

wty@wty-YangTianT4900k-00IRB:~/workspace/ros_ur_driver$ ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5 robot_ip:=192.168.56.101 launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/wty/.ros/log/2025-03-21-16-16-31-129915-wty-YangTianT4900k-00IRB-132376
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [132380]
[INFO] [controller_stopper_node-4]: process started with pid [132384]
[INFO] [ur_ros2_control_node-1]: process started with pid [132378]
[INFO] [robot_state_helper-3]: process started with pid [132382]
[INFO] [urscript_interface-5]: process started with pid [132386]
[INFO] [robot_state_publisher-6]: process started with pid [132388]
[INFO] [rviz2-7]: process started with pid [132390]
[INFO] [spawner-8]: process started with pid [132392]
[INFO] [spawner-9]: process started with pid [132394]
[robot_state_publisher-6] [INFO] [1742544991.380870037] [robot_state_publisher]: got segment base
[robot_state_publisher-6] [INFO] [1742544991.380940300] [robot_state_publisher]: got segment base_link
[robot_state_publisher-6] [INFO] [1742544991.380945889] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-6] [INFO] [1742544991.380949023] [robot_state_publisher]: got segment flange
[robot_state_publisher-6] [INFO] [1742544991.380951662] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-6] [INFO] [1742544991.380954329] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-6] [INFO] [1742544991.380956875] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-6] [INFO] [1742544991.380959588] [robot_state_publisher]: got segment tool0
[robot_state_publisher-6] [INFO] [1742544991.380962030] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-6] [INFO] [1742544991.380964566] [robot_state_publisher]: got segment world
[robot_state_publisher-6] [INFO] [1742544991.380966917] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-6] [INFO] [1742544991.380969272] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-6] [INFO] [1742544991.380971543] [robot_state_publisher]: got segment wrist_3_link
[dashboard_client-2] [INFO] [1742544991.386713203] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[controller_stopper_node-4] [INFO] [1742544991.387312802] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[ur_ros2_control_node-1] [WARN] [1742544991.388084943] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ur_ros2_control_node-1] text not specified in the tf_prefix tag
[ur_ros2_control_node-1] [INFO] [1742544991.388368673] [resource_manager]: Loading hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1742544991.391100969] [resource_manager]: Initialize hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1742544991.391155699] [resource_manager]: Successful initialization of hardware 'ur5'
[ur_ros2_control_node-1] [INFO] [1742544991.391326386] [resource_manager]: 'configure' hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1742544991.391331887] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1742544991.391340849] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1742544991.391559297] [UR_Client_Library:]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
[ur_ros2_control_node-1] [INFO] [1742544991.392740439] [UR_Client_Library:]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1742544991.392816197] [UR_Client_Library:]: Setting up RTDE communication with frequency 125.000000
[spawner-9] [INFO] [1742544991.646131702] [spawner_joint_trajectory_controller]: waiting for service /controller_manager/list_controllers to become available...
[spawner-8] [INFO] [1742544991.659337710] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[rviz2-7] [INFO] [1742544991.784803277] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-7] [INFO] [1742544991.784903093] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-7] [INFO] [1742544991.831414331] [rviz2]: Stereo is NOT SUPPORTED
[ur_ros2_control_node-1] [INFO] [1742544994.449372793] [URPositionHardwareInterface]: Calibration checksum: 'calib_209549117540498681'.
[ur_ros2_control_node-1] [INFO] [1742544995.490049516] [URPositionHardwareInterface]: Calibration checked successfully.
[ur_ros2_control_node-1] [INFO] [1742544995.490119839] [URPositionHardwareInterface]: System successfully started!
[ur_ros2_control_node-1] [INFO] [1742544995.490134529] [resource_manager]: Successful 'configure' of hardware 'ur5'
[ur_ros2_control_node-1] [INFO] [1742544995.490180133] [resource_manager]: 'activate' hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1742544995.490185613] [URPositionHardwareInterface]: Activating HW interface
[ur_ros2_control_node-1] [INFO] [1742544995.490191282] [resource_manager]: Successful 'activate' of hardware 'ur5'
[controller_stopper_node-4] [INFO] [1742544995.492889562] [Controller stopper]: Service available
[controller_stopper_node-4] [INFO] [1742544995.492911775] [Controller stopper]: Waiting for list controllers service to come up on controller_manager/list_controllers
[controller_stopper_node-4] [INFO] [1742544995.492926157] [Controller stopper]: Service available
[ur_ros2_control_node-1] [WARN] [1742544995.494434428] [controller_manager]: Could not enable FIFO RT scheduling policy
[ur_ros2_control_node-1] [WARN] [1742544995.494550644] [UR_Client_Library:]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
[ur_ros2_control_node-1] [INFO] [1742544995.652561627] [controller_manager]: Loading controller 'joint_trajectory_controller'
[ur_ros2_control_node-1] [WARN] [1742544995.658174674] [joint_trajectory_controller]: [Deprecated]: "allow_nonzero_velocity_at_trajectory_end" is set to true. The default behavior will change to false.
[spawner-9] [INFO] [1742544995.663217648] [spawner_joint_trajectory_controller]: Loaded joint_trajectory_controller
[... 45 lines cut ...]
[spawner-8] [INFO] [1742544995.886996351] [spawner_joint_state_broadcaster]: Loaded tcp_pose_broadcaster
[ur_ros2_control_node-1] [INFO] [1742544995.887166687] [controller_manager]: Configuring controller 'tcp_pose_broadcaster'
[spawner-8] [INFO] [1742544995.911076257] [spawner_joint_state_broadcaster]: Configured and activated tcp_pose_broadcaster
[ur_ros2_control_node-1] [INFO] [1742544995.911934910] [controller_manager]: Loading controller 'ur_configuration_controller'
[spawner-8] [INFO] [1742544995.918995830] [spawner_joint_state_broadcaster]: Loaded ur_configuration_controller
[ur_ros2_control_node-1] [INFO] [1742544995.919144524] [controller_manager]: Configuring controller 'ur_configuration_controller'
[spawner-8] [INFO] [1742544995.943077694] [spawner_joint_state_broadcaster]: Configured and activated ur_configuration_controller
[ur_ros2_control_node-1] [INFO] [1742544995.943978612] [controller_manager]: Loading controller 'scaled_joint_trajectory_controller'
[... cut ...]
[ur_ros2_control_node-1] [INFO] [1742544996.111756865] [controller_manager]: Switch controller timeout is set to 0, using default 1s!
[ur_ros2_control_node-1] [INFO] [1742545003.805560723] [UR_Client_Library:]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1742545003.805628436] [UR_Client_Library:]: Sent program to robot
[ur_ros2_control_node-1] [INFO] [1742545014.027802082] [UR_Client_Library:]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1742545014.027862127] [UR_Client_Library:]: Sent program to robot

urfeex, 2025-03-21:
Hi, thank you for reaching out. As far as I understand you

- Have setup the robot and driver
- You created a program on the teach pendant containing the External Control node
- You start that program on the robot
- Nothing happens. **No error on the teach pendant?**
- The ROS driver shows the output listed above

At this stage the robot program tries to connect to the ROS driver on port 50001. Could you tell us a bit how you start the ROS driver? Are you using a native Linux system, WSL2, Docker, VM setup? 

As the robot seems to be connecting to 50002 fine (for receiving the script code) the IP addresses seem to be configured correctly. Is there maybe a firewall running that allows access on port 50002, but not on port 50001?

leoCUHKSZ, 2025-03-21:
Thanks for your reply, it is indeed a firewall problem, I did not notice that the port of 50001 was blocked by the firewall before, I use a linux system. As I am still new in the ros, this problem has been bothering me for many days, thank you for providing a solution so quickly.

closed 2025-03-21 by leoCUHKSZ