UniversalRobots/Universal_Robots_ROS2_Driver #1317: Driver does not work with custom URDF file

pythonus-nation, 2025-03-26:
### Affected ROS2 Driver version(s)

3.1.1

### Used ROS distribution.

Rolling

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

Real robot

### Robot SW / URSim version(s)

3.15.8.106339

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
The driver throws an error when trying to launch it with a custom URDF file.

## Issue details
We are trying to launch the ur driver with a custom URDF file. The file is attached. For some reason GitHub does not allow to upload .urdf files which is the reason why the file's extension is renamed to .txt.  Directly after the launch an error is thrown which we are not able to understand. The log output is pasted into the corresponding field.

## Steps to Reproduce
Launch the driver with this command: `ros2 launch ur_control.launch.py ur_type:= "ur5" robot_ip=: "<robot-ip>" description_file=:"<path-to-custom-urdf>/ur5.urdf"` 

## Expected Behavior
The driver starts without throwing errors.

## Actual Behavior
The mentioned above occurs.

[ur5.txt](https://github.com/user-attachments/files/19469175/ur5.txt)

### Relevant log output

```shell
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-3]: process started with pid [628689]
[INFO] [controller_stopper_node-5]: process started with pid [628691]
[INFO] [rviz_viewer.py-1]: process started with pid [628687]
[INFO] [ros2_control_node-2]: process started with pid [628688]
[INFO] [robot_state_helper-4]: process started with pid [628690]
[INFO] [urscript_interface-6]: process started with pid [628692]
[INFO] [robot_state_publisher-7]: process started with pid [628693]
[INFO] [rviz2-8]: process started with pid [628694]
[INFO] [spawner-9]: process started with pid [628695]
[INFO] [spawner-10]: process started with pid [628708]
[controller_stopper_node-5] [INFO] [1742998977.041408227] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[dashboard_client-3] [INFO] [1742998977.047059042] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-3] 
[ros2_control_node-2] [INFO] [1742998977.162691190] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[robot_state_publisher-7] [INFO] [1742998977.184259457] [robot_state_publisher]: Robot initialized
[ros2_control_node-2] [INFO] [1742998977.190750107] [controller_manager]: update rate is 125 Hz
[ros2_control_node-2] [INFO] [1742998977.190818784] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-2] [WARN] [1742998977.191618728] [controller_manager]: Could not enable FIFO RT scheduling policy: with error number <1>(Operation not permitted). See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[ros2_control_node-2] [INFO] [1742998977.309350590] [controller_manager]: Received robot description from topic.
[ros2_control_node-2] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-2]   what():  no ros2_control tag
[ros2_control_node-2] Stack trace (most recent call last):
[ros2_control_node-2] #20   Object "", at 0xffffffffffffffff, in 
[ros2_control_node-2] #19   Object "/opt/ros/jazzy/lib/controller_manager/ros2_control_node", at 0x6124635d4ef4, in _start
[ros2_control_node-2] #18   Source "../csu/libc-start.c", line 360, in __libc_start_main_impl [0x75eda142a28a]
[ros2_control_node-2] #17   Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x75eda142a1c9]
[ros2_control_node-2] #16   Object "/opt/ros/jazzy/lib/controller_manager/ros2_control_node", at 0x6124635d3e94, in main
[ros2_control_node-2] #15   Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x75eda1d190e3, in rclcpp::executors::MultiThreadedExecutor::spin()
[ros2_control_node-2] #14   Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x75eda1d18dc6, in rclcpp::executors::MultiThreadedExecutor::run(unsigned long)
[ros2_control_node-2] #13   Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x75eda1d07c29, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
[ros2_control_node-2] #12   Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x75eda1d074fa, in rclcpp::Executor::execute_subscription(std::shared_ptr<rclcpp::SubscriptionBase>)
[ros2_control_node-2] #11   Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x75eda2058fd4, in 
[ros2_control_node-2] #10   Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x75eda1fe56ea, in controller_manager::ControllerManager::robot_description_callback(std_msgs::msg::String_<std::allocator<void> > const&)
[ros2_control_node-2] #9    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x75eda1fe4eae, in controller_manager::ControllerManager::init_resource_manager(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ros2_control_node-2] #8    Object "/opt/ros/jazzy/lib/libhardware_interface.so", at 0x75eda1af86ee, in hardware_interface::ResourceManager::load_and_initialize_components(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int)
[ros2_control_node-2] #7    Object "/opt/ros/jazzy/lib/libhardware_interface.so", at 0x75eda1ab803d, in 
[ros2_control_node-2] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33", at 0x75eda18bb390, in __cxa_throw
[ros2_control_node-2] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33", at 0x75eda18a5a54, in std::terminate()
[ros2_control_node-2] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33", at 0x75eda18bb0d9, in 
[ros2_control_node-2] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33", at 0x75eda18a5ff4, in 
[ros2_control_node-2] #2    Source "./stdlib/abort.c", line 79, in abort [0x75eda14288fe]
[ros2_control_node-2] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x75eda144527d]
[ros2_control_node-2] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[ros2_control_node-2]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[ros2_control_node-2]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x75eda149eb2c]
[ros2_control_node-2] Aborted (Signal sent by tkill() 628688 1000)
[rviz2-8] [INFO] [1742998977.624737076] [rviz2]: Stereo is NOT SUPPORTED
```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-03-26:
As your log output states, your URDF doesn't contain a `ros2_control` tag. In the driver we assemble a description from `ur_description` and a [ros2_control description](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/urdf/ur.ros2_control.xacro) into a [complete description](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/urdf/ur.urdf.xacro). This is the one passed to the [ur_rsp.launch.py](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/launch/ur_rsp.launch.py).

One way to integrate a custom URDF with that structure is explained in our [tutorial on a custom workcell](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_tutorials/my_robot_cell/doc/start_ur_driver.html).

I hope, that helps.

pythonus-nation, 2025-03-26:
Thanks for link! We will check it out. Is it generally possible to pass a urdf file to the driver or must it be a .xacro? If we can pass a urdf we would like to convert the xacro file into a urdf an then inlcude the content into our custom urdf.

urfeex, 2025-03-27:
It can definitively be a urdf file but from a maintainability perspective it seems much easier to keep xacro files and assembling the urdf on the fly (the xacro tool does exactly that).

pythonus-nation, 2025-03-28:
Ok, thanks for that info. We will try to create a correct URDF file today. If we run into new problems we will create a new post here.

closed 2025-04-23 by pythonus-nation