UniversalRobots/Universal_Robots_ROS2_Driver #1319: UR5 connected with external control but not moving

jtarulli, 2025-03-27:
### Affected ROS2 Driver version(s)

Ros2 humble 

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

Real robot

### Robot SW / URSim version(s)

UR5

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

---

# **Summary**  
Our UR5 robot suddenly stopped moving when sending trajectories via `scaled_joint_trajectory_controller`. It **worked perfectly 10 days ago**, and we changed **nothing** in the setup. Everything looks fine—no errors, the controller is active, commands are being published—but **the robot does not move**.  

## **Issue details**  
We are using UR5 with ROS 2 Humble and `Universal_Robots_ROS2_Driver`. The setup looks **correct**:  
- The robot is connected and **updates live in RViz** when moved manually in Freedrive mode.  
- `ros2 control list_controllers` confirms that `scaled_joint_trajectory_controller` is **active**.  
- We publish a valid `JointTrajectory` message to `/scaled_joint_trajectory_controller/joint_trajectory`.  
- The message appears in `ros2 topic echo`, but **nothing happens**.  

This **used to work perfectly**, and now it’s completely unresponsive.  

## **Steps to Reproduce**  
1. Start the robot:  
   ```bash
   ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.1.10
   ```  
2. Verify controllers:  
   ```bash
   ros2 control list_controllers
   ```  
   Output shows `scaled_joint_trajectory_controller` is **active**.  
3. Send a trajectory:  
   ```python
   from trajectory_msgs.msg import JointTrajectory, JointTrajectoryPoint
   import rclpy
   from rclpy.node import Node
   from builtin_interfaces.msg import Duration
   
   rclpy.init()
   node = rclpy.create_node('test_trajectory')
   pub = node.create_publisher(JointTrajectory, '/scaled_joint_trajectory_controller/joint_trajectory', 10)
   
   traj = JointTrajectory()
   traj.joint_names = ["shoulder_pan_joint", "shoulder_lift_joint", "elbow_joint", "wrist_1_joint", "wrist_2_joint", "wrist_3_joint"]
   
   point = JointTrajectoryPoint()
   point.positions = [0.0, -1.0, 1.0, -1.5, 0.5, 0.0]
   point.time_from_start = Duration(sec=2)
   
   traj.points.append(point)
   pub.publish(traj)
   ```  
4. **Result**: The message is published, but the **robot does not move**.  

## **Expected Behavior**  
The robot should execute the trajectory as it did **10 days ago**.  

## **Actual Behavior**  
Nothing happens. The robot is just **ignoring the command**, no errors, no feedback, nothing.  

## **Workaround Suggestions**  
We’ve already checked:  
✅ The robot is **not in Freedrive mode**.  
✅ The controller is **active** and receiving commands.  
✅ **RViz updates correctly** when moving the robot manually.  
✅ **No conflicts** between controllers.  
✅ The trajectory message is correctly formatted.  

We’ve tried **everything**, and we are out of ideas. **PLEASE HELP!** Has anyone faced this issue before?

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-03-28:
Could you please post the complete log output from 

```
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.1.10
```

jtarulli, 2025-03-28:
> Could you please post the complete log output from
> 
> ```
> ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.1.10
> ```

I got this output: 

(I mispelled the robot’s name, we’re using an UR5)

```
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5 robot_ip:=192.168.1.102 launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/ubunutulinux/.ros/log/2025-03-28-11-19-38-213382-ubunutulinuxMatteo-3582
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [3586]
[INFO] [controller_stopper_node-3]: process started with pid [3588]
[INFO] [ur_ros2_control_node-1]: process started with pid [3584]
[INFO] [urscript_interface-4]: process started with pid [3590]
[INFO] [robot_state_publisher-5]: process started with pid [3592]
[INFO] [rviz2-6]: process started with pid [3594]
[INFO] [spawner-7]: process started with pid [3596]
[INFO] [spawner-8]: process started with pid [3598]
[rviz2-6] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[robot_state_publisher-5] [INFO] [1743157178.545289657] [robot_state_publisher]: got segment base
[robot_state_publisher-5] [INFO] [1743157178.545485364] [robot_state_publisher]: got segment base_link
[robot_state_publisher-5] [INFO] [1743157178.545510076] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-5] [INFO] [1743157178.545526829] [robot_state_publisher]: got segment flange
[robot_state_publisher-5] [INFO] [1743157178.545541907] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-5] [INFO] [1743157178.545556827] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-5] [INFO] [1743157178.545571528] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-5] [INFO] [1743157178.545586358] [robot_state_publisher]: got segment tool0
[robot_state_publisher-5] [INFO] [1743157178.545600403] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-5] [INFO] [1743157178.545615231] [robot_state_publisher]: got segment world
[robot_state_publisher-5] [INFO] [1743157178.545630916] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-5] [INFO] [1743157178.545646319] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-5] [INFO] [1743157178.545661415] [robot_state_publisher]: got segment wrist_3_link
[ur_ros2_control_node-1] [WARN] [1743157178.553318142] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ur_ros2_control_node-1] text not specified in the tf_prefix tag
[ur_ros2_control_node-1] [INFO] [1743157178.554287700] [resource_manager]: Loading hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1743157178.558827229] [resource_manager]: Initialize hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1743157178.558977618] [resource_manager]: Successful initialization of hardware 'ur5'
[ur_ros2_control_node-1] [INFO] [1743157178.559418581] [resource_manager]: 'configure' hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1743157178.559446323] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1743157178.559476176] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1743157178.561568044] [UR_Client_Library:]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
[controller_stopper_node-3] [INFO] [1743157178.564219871] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[dashboard_client-2] [INFO] [1743157178.568283887] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[spawner-7] [INFO] [1743157178.882319078] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[rviz2-6] [INFO] [1743157178.914145970] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-6] [INFO] [1743157178.914292720] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[spawner-8] [INFO] [1743157178.929064851] [spawner_joint_trajectory_controller]: waiting for service /controller_manager/list_controllers to become available...
[rviz2-6] [INFO] [1743157178.932698617] [rviz2]: Stereo is NOT SUPPORTED
[ur_ros2_control_node-1] [INFO] [1743157178.981749477] [UR_Client_Library:]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1743157178.982423843] [UR_Client_Library:]: Setting up RTDE communication with frequency 125.000000
[ur_ros2_control_node-1] [INFO] [1743157180.099225767] [URPositionHardwareInterface]: Calibration checksum: 'calib_209549117540498681'.
[ur_ros2_control_node-1] [ERROR] [1743157181.113916936] [URPositionHardwareInterface]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_calibration/README.md] for details.
[... cut ...]
[ur_ros2_control_node-1] [INFO] [1743157182.210443626] [scaled_joint_trajectory_controller]: Controller state will be published at 100.00 Hz.
[ur_ros2_control_node-1] [INFO] [1743157182.213110730] [scaled_joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[spawner-7] [INFO] [1743157182.240609191] [spawner_joint_state_broadcaster]: Configured and activated scaled_joint_trajectory_controller
[INFO] [spawner-7]: process has finished cleanly [pid 3596]
[ur_ros2_control_node-1] [INFO] [1743157182.410478505] [controller_manager]: Switch controller timeout is set to 0, using default 1s!
```

urfeex, 2025-03-28:
Your output doesn't show a connection from the robot. That looks like [this item](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/usage/toc.html#i-started-everything-but-i-cannot-control-the-robot) from our documentation's troubleshooting section. With this output, I honestly doubt, that `ros2 control list_controllers` shows the scaled JTC as active, unless you manually started it while the program was not running.

Did you start the program on the robot's teach pendant?

When I run the driver, press play on the teach pendant and execute your script, the robot moves.

If I may add a suggestion: It would be better to use the controller's action interface than directly publishing to the `joint_trajectory` topic. This way, you will have proper state handling. See our [example_move.py](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/scripts/example_move.py) script for details.

jtarulli, 2025-03-31:
Now we started the driver and then the external control program on the Teach Pendant.

That's the output we obtained:

```
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5 robot_ip:=192.168.1.102 launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/ubunutulinux/.ros/log/2025-03-31-10-52-47-905169-ubunutulinuxMatteo-3467
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [3471]
[INFO] [controller_stopper_node-4]: process started with pid [3475]
[INFO] [ur_ros2_control_node-1]: process started with pid [3469]
[INFO] [robot_state_helper-3]: process started with pid [3473]
[INFO] [urscript_interface-5]: process started with pid [3477]
[INFO] [robot_state_publisher-6]: process started with pid [3479]
[INFO] [rviz2-7]: process started with pid [3481]
[INFO] [spawner-8]: process started with pid [3483]
[INFO] [spawner-9]: process started with pid [3485]
[dashboard_client-2] [INFO] [1743411171.205875909] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[controller_stopper_node-4] [INFO] [1743411171.372176600] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[rviz2-7] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[robot_state_publisher-6] [INFO] [1743411171.702341518] [robot_state_publisher]: got segment base
[robot_state_publisher-6] [INFO] [1743411171.702797489] [robot_state_publisher]: got segment base_link
[robot_state_publisher-6] [INFO] [1743411171.702836665] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-6] [INFO] [1743411171.702862064] [robot_state_publisher]: got segment flange
[robot_state_publisher-6] [INFO] [1743411171.702884913] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-6] [INFO] [1743411171.702908850] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-6] [INFO] [1743411171.702930649] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-6] [INFO] [1743411171.702952251] [robot_state_publisher]: got segment tool0
[robot_state_publisher-6] [INFO] [1743411171.702973331] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-6] [INFO] [1743411171.702995006] [robot_state_publisher]: got segment world
[robot_state_publisher-6] [INFO] [1743411171.703015837] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-6] [INFO] [1743411171.703036784] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-6] [INFO] [1743411171.703057473] [robot_state_publisher]: got segment wrist_3_link
[ur_ros2_control_node-1] [WARN] [1743411173.052350894] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ur_ros2_control_node-1] text not specified in the tf_prefix tag
[ur_ros2_control_node-1] [INFO] [1743411173.053506874] [resource_manager]: Loading hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1743411173.261327123] [resource_manager]: Initialize hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1743411173.261404391] [resource_manager]: Successful initialization of hardware 'ur5'
[ur_ros2_control_node-1] [INFO] [1743411173.261598225] [resource_manager]: 'configure' hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1743411173.261606112] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1743411173.261616466] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1743411173.315262782] [UR_Client_Library:]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
[ur_ros2_control_node-1] [INFO] [1743411173.324450030] [UR_Client_Library:]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1743411173.325171763] [UR_Client_Library:]: Setting up RTDE communication with frequency 125.000000
[spawner-9] [INFO] [1743411173.529166608] [spawner_joint_trajectory_controller]: waiting for service /controller_manager/list_controllers to become available...
[spawner-8] [INFO] [1743411173.537184329] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[ur_ros2_control_node-1] [INFO] [1743411174.436679514] [URPositionHardwareInterface]: Calibration checksum: 'calib_209549117540498681'.
[rviz2-7] [INFO] [1743411175.207698052] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-7] [INFO] [1743411175.207894130] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-7] [INFO] [1743411175.375731787] [rviz2]: Stereo is NOT SUPPORTED
[ur_ros2_control_node-1] [ERROR] [1743411175.492910181] [URPositionHardwareInterface]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_calibration/README.md] for details.
[... cut ...]
[INFO] [1743411914.065307042] [robot_controller]: Posizioni dei giunti: [0.143128, -1.18824, 1.4717900000000002, -1.7220799999999998, -1.53632, -0.07999999999999999]
[INFO] [1743411914.065712878] [robot_controller]: Invio della traiettoria al robot.
[INFO] [1743411914.066227815] [robot_controller]: Inviata la traiettoria. In attesa di completamento...
[INFO] [1743411917.069813244] [robot_controller]: Movimento completato!
```

The output seems correct but the robot doesn't move.

By the way this is the Teach command variables when we run the driver.
[image]

urfeex, 2025-03-31:
There is no `Robot connected to reverse interface. Ready to receive control commands.` in our output. That means the robot isn't connected to the driver correctly. Two possible reasons come to my mind:

- There is a firewall on the ROS machine preventing the robot to connect to the driver on port 50001 (You need to have ports 50001-50004 available).
- In case you have a container or VM setup running, port forwarding could be problematic in the same way as with a firewall.

jtarulli, 2025-03-31:
How can we verify the connection of that driver? 

Anyway, we are still connected on port 50002, the same port that worked fine until last week. We aren't running this on a VM, we're using a native Ubuntu 22.04 so we don't think that's a firewall problem. How can we try to fix this? Do you have any command to check?

We also noted an error while running the driver about the calibration, whit this output:

```
[ur_ros2_control_node-1] [ERROR] [1743411175.492910181] [URPositionHardwareInterface]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_calibration/README.md] for details.
```

it's an important error or we can pass through it? Thanks for the help

urfeex, 2025-03-31:
> We also noted an error while running the driver about the calibration, whit this output:
>
> ```
> [ur_ros2_control_node-1] [ERROR] [1743411175.492910181] [URPositionHardwareInterface]: The calibration [...]

You can ignore that for now. See the [calibration extraction](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/installation/robot_setup.html#calibration-extraction) and [verification](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/usage/startup.html#verify-calibration-info-is-being-used-correctly) for details on that.

To debug things, you can do the following

1. Start the ROS driver
2. SSH to the robot controller `ssh root@192.168.1.102`, pw `easybot` or whatever you have changed it to.
3. Type `nc <your-ros-machine-ip-address> 50001`
4. You should see the `Robot connected to reverse interface. Ready to receive control commands.` in your driver's output. **If not, stop here**. There's something standing in the way of the robot connecting to the driver.
5. Press CTRL-C on the ssh session. The driver should print `Connection to reverse interface dropped.`
6. On the teach pendant go into the program mode and load your program containing the external control program node. Modify your program e.g. by adding a PopUp node to the program.
7. Save the program. This will generate a couple of files on the robot one of which we will download next.
8. From a shell on your ROS machine run `scp root@192.168.1.102:/programs/<your-program-name>.script .` to copy the generated script code to your local machine. Upload that script code here.

jtarulli, 2025-04-02:
We did every step but during the 4th no output happened. What sould we try next? 

By the way, every ping on our ip seems to work.

closed 2025-04-03 by urfeex