UniversalRobots/Universal_Robots_ROS2_Driver #1329: External Error Control: after using move_group_interface.stop()

herreraverajosem, 2025-04-01:
### Affected ROS2 Driver version(s)

2.3.12 (ur_robot_driver)

### Used ROS distribution.

Iron

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

URSim in docker

### Robot SW / URSim version(s)

UR20/5.21.0

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
The goal is to send trajectories to URSim and be able to make an immediate stop at a certain moment, then resume the execution of the trajectory. The problem is that after using the **move_group_interface.stop()** function, URSim displays two errors and disconnects from the controller.

## Issue details
The problem is that after using the **move_group_interface.stop()** function, URSim displays two errors and disconnects from the controller. The goal is to immediately stop a trajectory in progress and then be able to resume it.

[image]
[image]

## Steps to Reproduce
The setup requires:

- URSim in Docker
- Ubuntu 22.04.5 Jammy LTS
- ROS 2 Iron
- Five nodes are needed:
  - ur_robot_driver node
  - MoveIt node
  - Node A: Sends trajectories to Node B and sends resume or stop signals to Node C.
  - Node B: Receives trajectories and executes them using move_group_interface.execute(cartesian_path) if the active signal is resume.
  - Node C: Receives stop or resume signals. In the case of a stop, it uses move_group_interface.stop() to halt the trajectory.

## Expected Behavior
It is expected that after stopping, a new trajectory can be sent, and the robotic arm movement will resume.

## Actual Behavior
After stopping, it is not possible to send a new trajectory and resume the robotic arm's movement; instead, URSim revokes control permissions.

# Workaround Suggestion
Is it possible to modify the speed or target of a trajectory in progress?

### Relevant log output

```shell
[ur_ros2_control_node-1] [INFO] [1743516996.011450834] [scaled_joint_trajectory_controller]: Goal reached, success!
[ur_ros2_control_node-1] [INFO] [1743516996.158375989] [scaled_joint_trajectory_controller]: Received new action goal
[ur_ros2_control_node-1] [INFO] [1743516996.158395586] [scaled_joint_trajectory_controller]: Accepted new action goal
[ur_ros2_control_node-1] [INFO] [1743516996.615097046] [scaled_joint_trajectory_controller]: Got request to cancel goal
[ur_ros2_control_node-1] [INFO] [1743516996.615121863] [scaled_joint_trajectory_controller]: Canceling active action goal because cancel callback received.
[ur_ros2_control_node-1] [INFO] [1743516996.868512180] [UR_Client_Library:]: Connection to reverse interface dropped.

[move_group-1] [INFO] [1743516996.158440771] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: scaled_joint_trajectory_controller started execution
[move_group-1] [INFO] [1743516996.158448276] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[move_group-1] [INFO] [1743516996.614931273] [moveit_ros.trajectory_execution_manager]: Received event 'stop'
[move_group-1] [INFO] [1743516996.614956671] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Cancelling execution for scaled_joint_trajectory_controller
[move_group-1] [INFO] [1743516996.615195963] [moveit_ros.trajectory_execution_manager]: Stopped trajectory execution.
[move_group-1] [INFO] [1743516996.658661804] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'scaled_joint_trajectory_controller' successfully finished
[move_group-1] [INFO] [1743516996.658708813] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status PREEMPTED ...
[move_group-1] [INFO] [1743516996.658772283] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: PREEMPTED
```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-04-01:
Thank you for reporting this. Please note: Iron is End of Life since a while now, so we cannot really answer any Iron-specific issues. Is this reproducible also on Humble?

The issue with the popup being too long has been fixed in the ur_client_library, but that will never land in Iron.

As far as I can see, the move_group simply cancels the trajectory, so we could remove the move_group from reproducing this in order to reduce complexity.

Does this happen with every motion you cancel?

urfeex, 2025-04-02:
We are not involved in any way with MoveIt, so I cannot give any help on writing programs with the move_group interface. If the original issue is resolved, then I will close this issue.

closed 2025-04-02 by urfeex