UniversalRobots/Universal_Robots_ROS2_Driver #1346: Universal Robots ROS2 Driver no longer builds successfully

rlgraf, 2025-04-16:
### Affected ROS2 Driver version(s)

newest

### Used ROS distribution.

Other

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

not sure

### How is the ROS driver used.

Others

### Issue details

When I build, there seems to be errors directly tied to the tool_contact_controller.hpp file. It is missing tool_contact.hpp. Below is the error:  

--- stderr: ur_controllers                             
In file included from /home/russell/ROS2_REPAS/src/Universal_Robots_ROS2_Driver/ur_controllers/src/tool_contact_controller.cpp:42:
/home/russell/ROS2_REPAS/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/tool_contact_controller.hpp:59:10: fatal error: ur_msgs/action/tool_contact.hpp: No such file or directory
   59 | #include <ur_msgs/action/tool_contact.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76: CMakeFiles/ur_controllers.dir/src/tool_contact_controller.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:409: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ur_controllers [0.64s, exited with code 2]

The Driver was building perfectly fine before recent updates, and in fact still works fine in its previous version on a virtual machine I have, before I recloned the updated driver over.

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-04-17:
If you're building the driver from the newest version (whatever that means, I guess the `main` branch?), please make sure to also keep the dependencies up-to-date.

The tool-contact action has been released in [ur_msgs 2.2.0](https://github.com/ros-industrial/ur_msgs/blob/humble-devel/CHANGELOG.rst#220-2025-03-17) and that has been synchronized to all supported ROS 2 distributions (you didn't specify yours)

Please consider installing the binary packages as explained in the [documentation](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/installation/installation.html#install-from-binary-packages) unless you have a specific reason to build the driver from source.

closed 2025-04-17 by urfeex

rlgraf, 2025-04-18:
Thanks for the reply! After cloning the driver repository, when I try to update the dependencies via: 

vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver-not-released.${ROS_DISTRO}.repos

I get the following error: 

/usr/bin/vcs:6: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import load_entry_point
usage: vcs import [-h] [--input FILE_OR_URL] [--force] [--shallow]
                  [--recursive] [--retry N] [--skip-existing] [--debug] [-w N]
                  [--repos]
                  [path]
vcs import: error: argument --input: can't open 'src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver-not-released.jazzy.repos': [Errno 2] No such file or directory: 'src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver-not-released.jazzy.repos'

It can't seem to find the .repos. Any ideas?

rlgraf, 2025-04-18:
Thank you for your reply! I will now be using binary packages. When I install using binary packages, everything builds correctly. However, I do not see Universal_Robots_ROS2_Driver in my directory. Is this correct? Where does it "exist"? How do packages in Universal_Robots_ROS2_Description and Universal_Robots_ROS2_GZ_Simulation "find" the driver? I'm fairly new here; thanks for your patience.

urfeex, 2025-04-22:
When installing things though the apt packages, they get installed into `/opt/ros/` as any other ROS package.