UniversalRobots/Universal_Robots_ROS2_Driver #1349: Real Time Kernel not possible for Ubuntu 22.04 and Ros2 Humble

becketps, 2025-04-27:
### Affected ROS2 Driver version(s)

Ros2 Humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

URSim in a virtual machine

### Robot SW / URSim version(s)

Polyscope 5

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
*Short introduction to the issue and how it impact you and why*
It is not possible to follow the mentioned steps here for ubuntu 22.04 with the standard Kernel 6.8.0 and this description https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/doc/installation/real_time.html

## Issue details
*Detailed description help us understand the problem. Code are welcome!*
There is no Kernel available for 6.8.0 and therefore I took the stabel LTS Version 5.15.179 https://git.kernel.org/pub/scm/linux/kernel/git/rt/linux-stable-rt.git which I only can build with

make -j $(getconf _NPROCESSORS_ONLN) deb-pkg

The result is an Error 
dpkg-buildpackage: Information: Host-Architektur amd64
 dpkg-source -i.git --before-build .
 debian/rules clean
rm -rf debian/files debian/linux-*
make clean
 dpkg-source -i.git -b .
dpkg-source: Information: Quellformat »1.0« wird verwendet
dpkg-source: Warnung: Quellverzeichnis »linux-5.15.179« lautet nicht <Quellpaket>-<Ursprungsversion> »linux-upstream-5.15.179-rt84«
dpkg-source: Warnung: .orig-Verzeichnisname linux-5.15.179.orig ist nicht <Paket>-<Ursprungsversion> (linux-upstream-5.15.179-rt84.orig erwünscht)
dpkg-source: Information: linux-upstream wird unter Benutzung des existierenden linux-upstream_5.15.179-rt84.orig.tar.gz gebaut
dpkg-source: Information: linux-upstream wird in linux-upstream_5.15.179-rt84-1.diff.gz gebaut
dpkg-source: Fehler: Änderung in vmlinux-gdb.py kann nicht dargestellt werden:
dpkg-source: Fehler:   neue Version ist Symlink auf /home/aut-laptop/rt_kernel_build/linux-5.15.179/scripts/gdb/vmlinux-gdb.py
dpkg-source: Fehler:   alte Version ist nicht existierend
dpkg-source: Warnung: Löschen der Datei .scmversion wird ignoriert
dpkg-source: Warnung: durch den Diff werden die folgenden Dateien der Originalautoren verändert: 
 .clang-format
 .cocciconfig
 .get_maintainer.ignore
 .mailmap
 CREDITS
 MAINTAINERS
 README
dpkg-source: Information: verwenden Sie das Format »3.0 (quilt)«, um separate und dokumentierte Änderungen an den Dateien der Originalautoren zu erhalten, siehe dpkg-source(1)
dpkg-source: Fehler: nicht darstellbare Änderungen an der Quelle
dpkg-buildpackage: Fehler: Unterprozess dpkg-source -i.git -b . lieferte Exitstatus 1
make[1]: *** [scripts/Makefile.package:77: deb-pkg] Fehler 1
make: *** [Makefile:1623: deb-pkg] Fehler 2

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

Load the Kernel mentioned above follow the description and try to build it. 

## Expected Behavior
*What did you expect and why?*

I expect that the building process for the kernel works properly or a work around can be found for ubuntu 22.04

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

# Workaround Suggestion
*If a workaround has been found, you are welcome to share it.*

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-04-28:
Thank you for reaching out to us. We will review out tutorial accordingly.

As a workaround, you can always try using the lowlatency kernel with which we've had great results in the past.

Another alternative might be to use [Ubuntu's realtime-kernel](https://ubuntu.com/real-time) directly. Note: This requires ubuntu pro access.

becketps, 2025-04-28:
> Thank you for reaching out to us. We will review out tutorial accordingly.
>
> As a workaround, you can always try using the lowlatency kernel with which we've had great results in the past.
>
> Another [...]

Thanks for the reply. I already run Ubuntu 22.04 LTS Pro and have already tried to enable the realtime kernel by pro enable realtime-kernel. But It is then enabled but at the next reboot the screen keeps black and I have to reset the system by timeshift to gain the usable status. Do you have a hint how I can avoid this? 

How do you think that I can make the workaround with the lowlatency kernel? I have already installed it. I also run the Polyscope 5 in a VM but am not able to controll the robot by TCP/IP connection via the Ros2 interface. With python on it's own it works with the manual but not with the ros2 interface. It is as if it is ignored. I have already installed the URCap for external control. 

Thanks in advance.

urfeex, 2025-04-29:
For trouble with the RT-Kernel through Ubuntu pro I suggest contacting Canonical, the Ubuntu maintainers, directly.

> How do you think that I can make the workaround with the lowlatency kernel?

You can install a lowlatency kernel by running

```
sudo apt install linux-image-lowlatency
```

Then, proceed with setting up the user privileges as explained in https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_client_library/doc/real_time.html#setup-user-privileges-to-use-real-time-scheduling. This allows your user to shedule threads with higher priority. We have [made some tests](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/60f08359f6484d1671e12bab6e01fc49eb0fadc8/ur_robot_driver/doc/real_time_benchmarking.md) with the lowlatency a while back which showed good results also compared to a full PREEMPT_RT patched kernel.

> I also tried running Polyscope 5 in a VM, but I’m not able to control the robot via the ROS 2 interface over TCP/IP.

What exactly does that mean? If you want to test against a simulated robot I would suggest using the Docker-based URSim as explained [here](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/usage/simulation.html).

> Using Python directly according to the manual works, but the ROS 2 interface seems to be ignored, even though I have installed the URCap for External Control.

This sounds a bit like the [first item in our troubleshooting section](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/usage/toc.html#i-started-everything-but-i-cannot-control-the-robot).

becketps, 2025-05-05:
I tested the command at this page after starting the rosl launch ur_robot_driver sucessfully with the rviz visualisiation: https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/usage/move.html#move-the-robot

ros2 launch ur_robot_driver test_scaled_joint_trajectory_controller.launch.py

That gave me the following output. How can I avoid that this happens without any movement?

[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [publisher_joint_trajectory_controller-1]: process started with pid [30656]
[publisher_joint_trajectory_controller-1] [INFO] [1746403855.840977080] [publisher_scaled_joint_trajectory_controller]: Goal "pos1" has definition trajectory_msgs.msg.JointTrajectoryPoint(positions=[0.785, -1.57, 0.785, 0.785, 0.785, 0.785], velocities=[], accelerations=[], effort=[], time_from_start=builtin_interfaces.msg.Duration(sec=4, nanosec=0))
[publisher_joint_trajectory_controller-1] [INFO] [1746403855.841917583] [publisher_scaled_joint_trajectory_controller]: Goal "pos2" has definition trajectory_msgs.msg.JointTrajectoryPoint(positions=[0.0, -1.57, 0.0, 0.0, 0.0, 0.0], velocities=[], accelerations=[], effort=[], time_from_start=builtin_interfaces.msg.Duration(sec=4, nanosec=0))
[publisher_joint_trajectory_controller-1] [INFO] [1746403855.842666659] [publisher_scaled_joint_trajectory_controller]: Goal "pos3" has definition trajectory_msgs.msg.JointTrajectoryPoint(positions=[0.0, -1.57, 0.0, 0.0, -0.785, 0.0], velocities=[], accelerations=[], effort=[], time_from_start=builtin_interfaces.msg.Duration(sec=4, nanosec=0))
[publisher_joint_trajectory_controller-1] [INFO] [1746403855.843452871] [publisher_scaled_joint_trajectory_controller]: Goal "pos4" has definition trajectory_msgs.msg.JointTrajectoryPoint(positions=[0.0, -1.57, 0.0, 0.0, 0.0, 0.0], velocities=[], accelerations=[], effort=[], time_from_start=builtin_interfaces.msg.Duration(sec=4, nanosec=0))
[publisher_joint_trajectory_controller-1] [INFO] [1746403855.843600737] [publisher_scaled_joint_trajectory_controller]: Publishing 4 goals on topic "/scaled_joint_trajectory_controller/joint_trajectory" every {wait_sec_between_publish} s
[publisher_joint_trajectory_controller-1] [WARN] [1746403855.844960381] [publisher_scaled_joint_trajectory_controller]: Starting point limits exceeded for joint shoulder_lift_joint !
[publisher_joint_trajectory_controller-1] [WARN] [1746403855.845136081] [publisher_scaled_joint_trajectory_controller]: Starting point limits exceeded for joint elbow_joint !
[publisher_joint_trajectory_controller-1] [WARN] [1746403855.845289807] [publisher_scaled_joint_trajectory_controller]: Starting point limits exceeded for joint wrist_1_joint !
[publisher_joint_trajectory_controller-1] [WARN] [1746403855.845440271] [publisher_scaled_joint_trajectory_controller]: Starting point limits exceeded for joint wrist_2_joint !
[publisher_joint_trajectory_controller-1] [WARN] [1746403855.845593623] [publisher_scaled_joint_trajectory_controller]: Starting point limits exceeded for joint shoulder_pan_joint !
[publisher_joint_trajectory_controller-1] [WARN] [1746403861.845123724] [publisher_scaled_joint_trajectory_controller]: Start configuration is not within configured limits!
[publisher_joint_trajectory_controller-1] [WARN] [1746403867.845032780] [publisher_scaled_joint_trajectory_controller]: Start configuration is not within configured limits!

urfeex, 2025-05-05:
You'll have to make sure that the robot is either close to the [starting_point_configuration](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/81f71bcd577c5d5f74d69b04000fdf3614d0f147/ur_robot_driver/config/test_goal_publishers_config.yaml#L27-L32) or disable the check in the config file.

becketps, 2025-05-17:
even with the file I placed in the folder which did not exist before the robot program always gives an error and nothing moves when I execute 

ros2 launch ur_robot_driver test_scaled_joint_trajectory_controller.launch.py

Error:
[publisher_joint_trajectory_controller-1] [WARN] [1747520156.441740138] [publisher_scaled_joint_trajectory_controller]: Start configuration is not within configured limits!
[publisher_joint_trajectory_controller-1] [WARN] [1747520162.440443226] [publisher_scaled_joint_trajectory_controller]: Start configuration is not within configured limits!

I also placed the robot in the home position executed the command again but nothing moves even when I see the output comming through. Do you know where this positions are located and why no movement is seen? 

[publisher_joint_trajectory_controller-1] [INFO] [1747520486.484506490] [publisher_scaled_joint_trajectory_controller]: Sending goal trajectory_msgs.msg.JointTrajectoryPoint(positions=[0.785, -1.57, 0.785, 0.785, 0.785, 0.785], velocities=[], accelerations=[], effort=[], time_from_start=builtin_interfaces.msg.Duration(sec=4, nanosec=0)).

Do you have another idea what to do? Is there a command list and parameters of all topics, services and actions available?

urfeex, 2025-05-19:
> but there the controller got stucked and does not continue.

What exactly does that mean? What is the controller's (driver's) output?

> I in general the robot also runable in isaac Sim?

Does this relate in anyway to this thread? Otherwise I would suggest not going sideways too much. As far as I know, the answer is [yes](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.universal_robots/docs/index.html).