UniversalRobots/Universal_Robots_ROS2_Driver #1352: Problem to control real UR10 with robotiq 2f-140

Cavalletta98, 2025-05-02:
Hi, I'm trying to control a real UR10e with a Robotiq 2F-140 gripper using ROS 2 Jazzy and MoveIt. The setup works fine without the gripper, but when I include the gripper, it fails to operate correctly.

I'm currently using the Remote Control URCap on Polyscope 5.19, along with the RS485 URCap, since the gripper is directly connected to the robot’s tool flange. The robot is running in local control mode using the Remote Program on Polyscope (i.e., I have to press "Play" to allow the UR ROS driver to communicate with the robot).

I'm using the gripper driver from this repository: [PickNikRobotics/ros2_robotiq_gripper](https://github.com/PickNikRobotics/ros2_robotiq_gripper). The gripper’s xacro file includes the plugin for a custom controller, which requires access to the USB interface exposed with the launch flag use_tool_communication:=true. This USB device should be available once the robot driver is fully running.

The problem seems to be that the gripper controller is being launched before the robot-driver node has established communication with the real hardware, so the required tool port isn’t available yet.

To work around this, I’ve created local copies of ur_control.launch.py and ur_rsp.launch.py so I can load my custom xacro and controller config. However, the issue persists.

What would be the recommended way to delay the gripper controller initialization until the robot communication is fully established? Is there a way to synchronize or sequence the launch of the gripper controller accordingly?

urfeex, 2025-05-05:
This is an interesting problem, yes. One way to solve this, could be to not automatically start the gripper component at startup. This can be achieved using the controller_manager's `hardware_components_initial_state` parameter. See the [ros2_control docs](https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/userdoc.html#parameters) for details.

Since the gripper driver seems to be [connecting to the serial interface](https://github.com/PickNikRobotics/ros2_robotiq_gripper/blob/12e623212e6891a5fcc9af94d67b07e640916394/robotiq_driver/src/hardware_interface.cpp#L160) on `on_configure`, starting the component in state `unconfigured` should help. Then, you can use the controller_manager's `set_hardware_component_state` service to start the gripper.

Cavalletta98, 2025-05-05:
Ok, i will try to figure out how to do so. In the meantime i have done some investigation and i found that the problem seems to be the serial comunication once i create the serial port via this command `ros2 run ur_robot_driver tool_communication.py --ros-args -p robot_ip:=192.168.56.101 -p device_name:=/tmp/ttyUR`. I did this steps:

- Run the command `ros2 run ur_robot_driver tool_communication.py --ros-args -p robot_ip:=192.168.56.101 -p device_name:=/tmp/ttyUR`
- Use mbpoll to test the comunication like `mbpoll -0 -a 9 -b 115200 -P none -t 3:hex -r 0 -c 10 /tmp/ttyUR`

I'm getting connection time out.
Do you know what will be the problem?

Cavalletta98, 2025-05-05:
to understand if this is the problem i performed these steps:

- launch the ur driver like this: `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10e robot_ip:=192.168.56.101 kinematics_params_file:=/home/my_robot_calibration.yaml use_tool_communication:=true tool_voltage:=24 headless_mode:=true`
- use mbpoll to read from the serial like this: `mbpoll -0 -a 9 -b 115200 -P none -t 3:hex -r 0 -c 10 /tmp/ttyUR`

getting this error as before:
`mbpoll 1.0-0 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright © 2015-2019 Pascal JEAN, https://github.com/epsilonrt/mbpoll
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus RTU
Slave configuration...: address = [9]
                        start reference = 0, count = 10
Communication.........: /tmp/ttyUR,     115200-8N1 
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, input register table

-- Polling slave 9... Ctrl-C to stop)
Read input register failed: Connection timed out
-- Polling slave 9... Ctrl-C to stop)
Read input register failed: Connection timed out
-- Polling slave 9... Ctrl-C to stop)
Read input register failed: Connection timed out
-- Polling slave 9... Ctrl-C to stop)
Read input register failed: Connection timed out
-- Polling slave 9... Ctrl-C to stop)
Read input register failed: Connection timed out
-- Polling slave 9... Ctrl-C to stop)
Read input register failed: Connection timed out
-- Polling slave 9... Ctrl-C to stop)
Read input register failed: Connection timed out
-- Polling slave 9... Ctrl-C to stop)
Read input register failed: Connection timed out
-- Polling slave 9... Ctrl-C to stop)
Read input register failed: Connection timed out
-- Polling slave 9... Ctrl-C to stop)
Read input register failed: Connection timed out
-- Polling slave 9... Ctrl-C to stop)
Read input register failed: Connection timed out
-- Polling slave 9... Ctrl-C to stop)
Read input register failed: Connection timed out
-- Polling slave 9... Ctrl-C to stop)
^C--- /tmp/ttyUR poll statistics ---
13 frames transmitted, 0 received, 12 errors, 100.0% frame loss`

Cavalletta98, 2025-05-05:
@urfeex i'm trying to implement what you suggested. i correctly put the gripper into unconfigured at startup but i'm getting errors about joint state broadcaster: 
`[ros2_control_node-1] [ERROR] [1746439989.111125509] [joint_state_broadcaster]: No state interfaces found to publish.
[ros2_control_node-1] [ERROR] [1746439989.111193532] [joint_state_broadcaster]: Error initializing joint data. JointStateBroadcaster will not run.
[ros2_control_node-1] [WARN] [1746439989.111225095] [joint_state_broadcaster]: Error occurred while doing error handling.
[ros2_control_node-1] [ERROR] [1746439989.111258720] [controller_manager]: After activation, controller 'joint_state_broadcaster' is in state 'unconfigured' (1), expected 'active' (3).
[ros2_control_node-1] [ERROR] [1746439989.111309623] [controller_manager]: Could not activate controller : 'joint_state_broadcaster'
[ros2_control_node-1] [ERROR] [1746439989.111344831] [controller_manager]: Failed switching controllers.... 
[ros2_control_node-1] [ERROR] [1746439989.111367425] [controller_manager]: Unable to activate controllers: [ joint_state_broadcaster ]
[ros2_control_node-1] [INFO] [1746439989.113470982] [controller_manager]: Configuring controller: 'forward_velocity_controller'
[ros2_control_node-1] [INFO] [1746439989.113791872] [forward_velocity_controller]: configure successful
[spawner-9] [ERROR] [1746439989.114097751] [spawner_joint_state_broadcaster]: Failed to activate controller : joint_state_broadcaster
[ros2_control_node-1] [INFO] [1746439989.116895164] [controller_manager]: Loading controller : 'forward_position_controller' of type 'position_controllers/JointGroupPositionController'`

urfeex, 2025-05-05:
At this point, I cannot say much without seeing the actual assembled URDF and the exact commands and configs used to launch things.

urfeex, 2025-05-05:
As far as I can see from the files, you didn't include a ros2_control tag for the UR arm anywhere. Hence, ros2_control only knows your gripper, which isn't started. Hence, the joint state broadcaster isn't finding any joints.

Cavalletta98, 2025-05-05:
where i should include this tag? i did not understand

urfeex, 2025-05-06:
In your `ur10e_urdf.xacro` you add the robot:

[image]

I usually don't post screenshots of code, but in that case there's no code to refer to...

But that is only giving you the kinematic description of the robot. We have been factoring the ros2_control description out of the URDF as the URDF is reused for situations where the ros2_control / driver parameters are irrelevant. Therefore, you'll have to add the ros2_control tag separately as in https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/81f71bcd577c5d5f74d69b04000fdf3614d0f147/ur_robot_driver/urdf/ur.urdf.xacro#L76-L104

urfeex, 2025-05-06:
Potentially, the gripper is slowing down the control loop. In that case you would probably need asynchronous hardware interfaces. See https://control.ros.org/rolling/doc/ros2_control/hardware_interface/doc/asynchronous_components.html for details.

Cavalletta98, 2025-05-06:
I was trying to understand how to perform it following your suggestion. i added the async flag in the ros2 control tag of the gripper as follow ( i will not add all the lines, just the most significant)
`<ros2_control name="${name}" type="system" is_async="true">
            
            <hardware>
                <xacro:if value="${sim_isaac}">
                    <plugin>topic_based_ros2_control/TopicBasedSystem</plugin>
                    <param name="joint_commands_topic">${isaac_joint_commands}</param>
                    <param name="joint_states_topic">${isaac_joint_states}</param>
                </xacro:if>
                <xacro:if value="${sim_gazebo}">
                    <plugin>ign_ros2_control/IgnitionSystem</plugin>
                </xacro:if>
                <xacro:if value="${use_fake_hardware}">
                    <plugin>mock_components/GenericSystem</plugin>
                    <param name="mock_sensor_commands">${mock_sensor_commands}</param>
                    <param name="state_following_offset">0.0</param>
                </xacro:if>
                <xacro:unless value="${use_fake_hardware or sim_gazebo or sim_isaac}">
                    <plugin>robotiq_driver/RobotiqGripperHardwareInterface</plugin>
                    <param name="gripper_closed_position">${gripper_closed_position}</param>
                    <param name="COM_port">${com_port}</param>
                    <param name="gripper_speed_multiplier">${gripper_speed_multiplier}</param>
                    <param name="gripper_force_multiplier">${gripper_force_multiplier}</param>
                    <param name="gripper_max_speed">${gripper_max_speed}</param>
                    <param name="gripper_max_force">${gripper_max_force}</param>
                </xacro:unless>
            </hardware>`
I'm still getting the same error. I have two doubts:

- Do you think that should be activated something in the gripper driver to use the async mode?
- Do you think that the fact that the `type=system` together with  `is_async="true` will not work? it is seems that, in the link shared by you, the async flag is no used together with type flag

urfeex, 2025-05-15:
Honestly, from this point on there's too much assumptions from my side, as I have personally never integrated a robotiq gripper with a UR arm in ROS 2. Hopefully, I will be able to do that soon, I will come back here if that happens.

macmacal, 2025-09-24:
Hi, I would like to add some more related information. I'm using ROS 2 Humble. During integration of a UR5e robot with Robotiq's Hand-E gripper via our package [robotiq_hande_driver](https://github.com/AGH-CEAI/robotiq_hande_driver) based on [libmodbus](https://libmodbus.org/) we have encounter problems with the "jerks" during trajectory execution. During every "jump" we logged the `Pipeline producer overflowed! <RTDE Data Pipeline>` error.

To localize the problem, we have connected the robot directly to the PC with Ubuntu 24.04 real-time kernel PREEMPT_RT and disabled all ros2_control modules, then enabling just the robot and our gripper driver. Without the gripper driver, the problem vanishes. 

Since I don't have much time for debugging it further, we are investigating the @Cavalletta98 suggestion of using external RS485 USB adapter.

urfeex, 2025-09-25:
> Without the gripper driver, the problem vanishes.

If the gripper driver also uses ros2_control and is running on the same controller_manager as the UR driver, then the issue might actually be with the gripper driver where it slows down the main control loop. If that happens, jerks and pipeline overflows are expected.