UniversalRobots/Universal_Robots_ROS2_Driver #1356: UR5e connected with external control but wont move

vhczz, 2025-05-09:
### Affected ROS2 Driver version(s)

Ros2 Jazzy

### Used ROS distribution.

Other

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

UR5e

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I am trying to load a python code to move the physical robot, I run the drivers, see the UR in Rviz and then try to run my launch file on another terminal. However when I run the code the robot wont move and the joint_trajectory_controller deactivates.

## Issue details
I also encounter some warning when running the drivers but overall it seem to be a problem with the controller (I think), however I should also mention when I ping the robot my latency tends to be quite high (around 6ms). Here is the output of the drivers on the bash: 
$ ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.1.135
[INFO] [launch]: All log files can be found below /home/victor-chavez/.ros/log/2025-05-08-13-46-29-693447-victor-chavezr-23296
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [23303]
[INFO] [controller_stopper_node-4]: process started with pid [23305]
[INFO] [ros2_control_node-1]: process started with pid [23302]
[INFO] [robot_state_helper-3]: process started with pid [23304]
[INFO] [urscript_interface-5]: process started with pid [23306]
[INFO] [robot_state_publisher-6]: process started with pid [23307]
[INFO] [rviz2-7]: process started with pid [23308]
[INFO] [spawner-8]: process started with pid [23309]
[INFO] [spawner-9]: process started with pid [23310]
[dashboard_client-2] [INFO] [1746733589.986122293] [UR_Client_Library:]: Starting primary client pipeline
[robot_state_publisher-6] [INFO] [1746733589.987576103] [robot_state_publisher]: Robot initialized
[ros2_control_node-1] [INFO] [1746733589.991340034] [controller_manager]: Using Steady (Monotonic) clock for triggering controller manager cycles.
[controller_stopper_node-4] [INFO] [1746733589.992111013] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[robot_state_helper-3] [INFO] [1746733589.993132302] [UR_Client_Library:]: Starting primary client pipeline
[ros2_control_node-1] [INFO] [1746733589.993954122] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[ros2_control_node-1] [INFO] [1746733589.997222486] [controller_manager]: update rate is 500 Hz
[ros2_control_node-1] [INFO] [1746733589.997265713] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-1] [WARN] [1746733589.997330136] [controller_manager]: Could not enable FIFO RT scheduling policy: with error number <1>(Operation not permitted). See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[dashboard_client-2] [INFO] [1746733590.053239214] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2]
[rviz2-7] [INFO] [1746733590.145002330] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-7] [INFO] [1746733590.145150204] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-7] [INFO] [1746733590.159727006] [rviz2]: Stereo is NOT SUPPORTED
[spawner-9] [INFO] [1746733590.305926183] [spawner_joint_trajectory_controller]: waiting for service /controller_manager/list_controllers to become available...
[spawner-8] [INFO] [1746733590.320894632] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[ros2_control_node-1] [WARN] [1746733590.993593289] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1746733591.993641595] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1746733592.993569804] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1746733593.993581980] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [INFO] [1746733594.568075638] [controller_manager]: Received robot description from topic.
[ros2_control_node-1] text not specified in the tf_prefix tag
[ros2_control_node-1] [INFO] [1746733594.570563610] [controller_manager]: Loading hardware 'ur5e'
[... 435 lines cut ...]
    rclpy.spin(node)
    rclpy.shutdown()

if __name__ == '__main__':
    main()

## Expected Behavior
I expect the physical robot to move after the python program is running, I also tried this same steps but with a simulation running ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e \
  robot_ip:=yyy.yyy.yyy.yyy use_mock_hardware:=true \
  initial_joint_controller:=joint_trajectory_controller

## Actual Behavior
After I reproduce these steps the physical robot won't move.

# Workaround Suggestion

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-05-13:
Hi, thank you for reaching out. I've taken the liberty to put your log output and python code into code blocks to make things more readable and I split the two different runs into two sections.

In the driver's log output you posted, I can see two starting attempts: In the first one the driver seems to take too much time to come up for the controller spawner to wait for it. That surprises me to be honest, as I wouldn't expect the driver to take longer than 10 seconds to start. It seems, your `controller_manager` receives the `robot_description` roughly 4 seconds after the `robot_state_publisher` has been initialized and then another 6 seconds later one spawner is happy with the `controller_manager` services, while the other dies. There seems to be an issue with your local ROS communication. I cannot really give you support on that, but one suggestion is to try using local discovery only, as explained [here](https://docs.ros.org/en/jazzy/Tutorials/Advanced/Improved-Dynamic-Discovery.html)

Your second attempt seems to be coming up correctly, but it appears that the robot program isn't started, so the robot won't accept any commands, hence the controller keeps being deactivated. See [our troubleshooting section](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/usage/toc.html#i-started-everything-but-i-cannot-control-the-robot) for details on that. You probably need to create a program on the robot as explained [here](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_client_library/doc/setup/robot_setup.html#urcap-installation) (Use the "PolyScope 5" tab).

Since your driver seems to be losing the connection to the robot rather often: Do you connect over a shared network with potentially high load? We suggest a direct connection between the ROS machine and the robot.

vhczz, 2025-05-14:
I am working with some other robots (some turtlebots4) where they are discovery servers. Although I don't think that's the problem because as you can see on the second attempt it seems to come up correctly but when I put the teach pendant on remote mode and load the program remote control program there then launch the code i get victor-chavez@victor-chavezr:~$ ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.1.51 kinematics_params_file:="/home/victor-chavez/my_robot_calibration.yaml"
[INFO] [launch]: All log files can be found below /home/victor-chavez/.ros/log/2025-05-13-19-58-09-488586-victor-chavezr-16181
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [16186]
[INFO] [controller_stopper_node-4]: process started with pid [16188]
[INFO] [ros2_control_node-1]: process started with pid [16185]
[INFO] [robot_state_helper-3]: process started with pid [16187]
[INFO] [urscript_interface-5]: process started with pid [16189]
[INFO] [robot_state_publisher-6]: process started with pid [16190]
[INFO] [rviz2-7]: process started with pid [16191]
[INFO] [spawner-8]: process started with pid [16192]
[INFO] [spawner-9]: process started with pid [16193]
[robot_state_helper-3] [INFO] [1747187889.962422297] [UR_Client_Library:]: Starting primary client pipeline
[dashboard_client-2] [INFO] [1747187889.966163159] [UR_Client_Library:]: Starting primary client pipeline
[controller_stopper_node-4] [INFO] [1747187889.968221735] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[robot_state_publisher-6] [INFO] [1747187889.978539160] [robot_state_publisher]: Robot initialized
[ros2_control_node-1] [INFO] [1747187890.002408051] [controller_manager]: Using Steady (Monotonic) clock for triggering controller manager cycles.
[ros2_control_node-1] [INFO] [1747187890.009157434] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[ros2_control_node-1] [INFO] [1747187890.013986218] [controller_manager]: update rate is 500 Hz
[ros2_control_node-1] [INFO] [1747187890.014031208] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-1] [WARN] [1747187890.014139124] [controller_manager]: Could not enable FIFO RT scheduling policy: with error number <1>(Operation not permitted). See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[dashboard_client-2] [INFO] [1747187890.068796326] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[rviz2-7] [INFO] [1747187890.286119726] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-7] [INFO] [1747187890.286292151] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-7] [INFO] [1747187890.310971246] [rviz2]: Stereo is NOT SUPPORTED
[spawner-9] [INFO] [1747187890.506844138] [spawner_joint_trajectory_controller]: waiting for service /controller_manager/list_controllers to become available...
[spawner-8] [INFO] [1747187890.512741652] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[ros2_control_node-1] [WARN] [1747187891.008778056] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1747187892.008752948] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1747187893.008767430] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1747187894.008753814] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [INFO] [1747187894.583478173] [controller_manager]: Received robot description from topic.
[ros2_control_node-1] text not specified in the tf_prefix tag
[ros2_control_node-1] [INFO] [1747187894.586854769] [controller_manager]: Loading hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1747187894.597007298] [controller_manager]: Loaded hardware 'ur5e' from plugin 'ur_robot_driver/URPositionHardwareInterface'
[ros2_control_node-1] [INFO] [1747187894.597152018] [controller_manager]: Initialize hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1747187894.597809515] [controller_manager]: Successful initialization of hardware 'ur5e'
[ros2_control_node-1] [INFO] [1747187894.600088375] [resource_manager]: 'configure' hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1747187894.600123560] [URPositionHardwareInterface]: Starting ...please wait...
[ros2_control_node-1] [INFO] [1747187894.600148230] [URPositionHardwareInterface]: Initializing driver...
[ros2_control_node-1] [WARN] [1747187894.606240360] [UR_Client_Library:]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_client_library/doc/real_time.html for details.
[ros2_control_node-1] [INFO] [1747187894.619832846] [UR_Client_Library:]: Negotiated RTDE protocol version to 2.
[ros2_control_node-1] [INFO] [1747187894.647094845] [UR_Client_Library:]: Setting up RTDE communication with frequency 500.000000
[ros2_control_node-1] [INFO] [1747187895.699874270] [UR_Client_Library:]: Starting primary client pipeline
[... cut ...]
[ros2_control_node-1] [ERROR] [1747187952.755086909] [scaled_joint_trajectory_controller]: Can't accept new action goals. Controller is not running.
[ros2_control_node-1] [WARN] [1747187952.855239214] [scaled_joint_trajectory_controller.rclcpp_action]: Failed to send goal response c30e5abd-f2b9-41e1-8498-db641d4e29da (timeout): client will not receive response, at ./src/rmw_response.cpp:153, at ./src/rcl/service.c:400
[ros2_control_node-1] [WARN] [1747187971.980764130] [UR_Client_Library:]: Failed to read from stream, reconnecting in 64 seconds...

urfeex, 2025-05-14:
I am still a bit worried about that `Failed to read from stream, reconnecting in 8 seconds...`. That indicates that you have temporarily lost connection to your robot. That's why I was asking about your network setup previously.

I can reproduce that behavior locally by unplugging the robot's network cable and plugging it back in. The RTDE client seems not to be reconnecting properly by its own. That's unwanted behavior, we will investigate.
As a side-effect the controller_stopper doesn't get notifier that the robot is running, so the controller is never started, which is what you see if you try to send it a trajectory.

Nevertheless: The fact that your control PC is losing connection to the robot is worrying and should be addressed. If you don't have a stable connection between your control PC and the robot, you will inevitably run into all sorts of problems.