UniversalRobots/Universal_Robots_ROS2_Driver #1377: Build failure in ur_controllers: missing tool_contact_controller_parameters.hpp

simonllopez, 2025-05-25:
### Affected ROS2 Driver version(s)

Humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

URSim in linux

### Robot SW / URSim version(s)

10e

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

When building the ur_controllers package on ROS 2 Humble, the build fails because the header tool_contact_controller_parameters.hpp cannot be found. After that, several compilation errors appear in scaled_joint_trajectory_controller.cpp related to the joint_trajectory_controller API.

Steps to Reproduce:

```
export COLCON_WS=~/workspace/ros_ur_driver

mkdir -p $COLCON_WS/src && cd $COLCON_WS

git clone -b humble https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git src/Universal_Robots_ROS2_Driver

vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver-not-released.humble.repos

rosdep update

rosdep install --ignore-src --from-paths src -y

colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release

```

### Relevant log output

```shell
--- stderr: ur_controllers
In file included from .../ur_controllers/src/tool_contact_controller.cpp:42:
/ur_controllers/include/ur_controllers/tool_contact_controller.hpp:60:10: fatal error: ur_controllers/tool_contact_controller_parameters.hpp: No such file or directory
   60 | #include "ur_controllers/tool_contact_controller_parameters.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
...
ur_controllers/src/scaled_joint_trajectory_controller.cpp:112:50: error: invalid initialization of reference of type ‘const joint_trajectory_controller::Params&’ from expression of type ‘rclcpp::Logger’
  112 |     default_tolerances_ = get_segment_tolerances(logger, params_);
      |                                                  ^~~~~~
...
ur_controllers/src/scaled_joint_trajectory_controller.cpp:203:25: error: ‘active_tolerances_’ was not declared in this scope
  203 |       auto active_tol = active_tolerances_.readFromRT();
      |                         ^~~~~~~~~~~~~~~~~~
...
gmake[2]: *** [CMakeFiles/ur_controllers.dir/...scaled_joint_trajectory_controller.cpp.o] Error 1
```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-05-26:
I was not able to reproduce this using a Humble Docker container:

``` 
$ docker run -it --rm ros:humble
root@6dde3143cae9:/# export COLCON_WS=~/workspace/ros_ur_driver && \
  mkdir -p $COLCON_WS/src && cd $COLCON_WS
  git clone -b humble https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git src/Universal_Robots_ROS2_Driver && \
  rosdep update && \
  rosdep install --ignore-src --from-paths src -y && \
  colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
[...]
Starting >>> ur_dashboard_msgs
Starting >>> ur_moveit_config
Finished <<< ur_moveit_config [0.74s]
Finished <<< ur_dashboard_msgs [5.30s]
Starting >>> ur_controllers
Finished <<< ur_controllers [11.3s]
Starting >>> ur_robot_driver
Starting >>> ur_bringup
Finished <<< ur_bringup [0.43s]
Finished <<< ur_robot_driver [12.3s]
Starting >>> ur_calibration
Finished <<< ur_calibration [8.30s]
Starting >>> ur
Finished <<< ur [0.45s]

Summary: 7 packages finished [37.8s]
```

Is your system up-to-date? The missing `ur_controllers/tool_contact_controller_parameters.hpp` indicates that there's something odd with your generate_parameters_library installation.

simonllopez, 2025-05-26:
It's working fine for me as well on a clean install. I'll look into how to update the part that's failing for me. Sorry for the trouble, and thanks for your help!

closed 2025-05-26 by urfeex