UniversalRobots/Universal_Robots_ROS2_Driver #1384: Direct torque control

ggorjup, 2025-05-29:
### Feature summary

Hello, UR recently announced a [new low-level torque interface](https://www.universal-robots.com/news-and-media/news-center/universal-robots-offers-new-low-level-torque-interface-to-empower-research-and-innovation/), which is currently in early access and available through a [beta version of Polyscope 5](https://ur.centercode.com/key/PolyScope5Beta).

As I understand, the hardware interface currently supports position and velocity commands. 
Is adding support for torque control feasible / on the roadmap?

urfeex, 2025-06-02:
We do indeed have it on our plate. There is a [beta branch in the client library](https://github.com/UniversalRobots/Universal_Robots_Client_Library/tree/torque_control) already and an implementation here in the driver will follow in the next weeks. I'll try to remember posting here once the PR is there.

urfeex, 2025-06-17:
Please see #1411 for a first implementation. That needs some polishing, hence the experimental state for now.

urfeex, 2025-10-15:
This got merged in #1411