UniversalRobots/Universal_Robots_ROS2_Driver #1399: ur_controllers build fails with ROS2 Jazzy

adkoessler, 2025-06-11:
### Affected ROS2 Driver version(s)

Jazzy

### Used ROS distribution.

Other

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.12

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
The ur_controllers package does not build from source. I tried to update everything. I can see from the frontpage that the build is failing. I tried checking out to former commits without any success.
How do I revert back to something that works? Any help is appreciated.

## Steps to Reproduce
update and upgrade apt
update and install rosdep
git clone https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git -b jazzy
colcon build

## Actual Behavior
See the log output.

### Relevant log output

```shell
Starting >>> ur_controllers
--- stderr: ur_controllers                             
In file included from /opt/ros/jazzy/include/controller_interface/controller_interface/controller_interface.hpp:21,
                 from /home/lcfcrobots/ur_workspace/src/Universal_Robots_ROS2_Driver/ur_controllers/src/freedrive_mode_controller.cpp:37:
/opt/ros/jazzy/include/controller_interface/controller_interface/controller_interface_base.hpp:26:10: fatal error: hardware_interface/introspection.hpp: No such file or directory
   26 | #include "hardware_interface/introspection.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-06-12:
@adkoessler That seems to be originating from a partially out-of-date system. The missing file in the `hardware_interface` package was [added in February](https://github.com/ros-controls/ros2_control/commits/master/hardware_interface/include/hardware_interface/introspection.hpp). Is your `hardware_interfaces` package up-to-date?

@Alanzelic Your issue seems to be completely unrelated. It looks like the `controller_interface` package isn't installed, at all. Did you run the `rosdep install ...` command [as explained in the docs](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/installation/installation.html#build-from-source)?

May I suggest to both of you to simply install the binary packages using `sudo apt-get install ros-jazzy-ur`? This is the recommended method of installing the driver unless you want to actually do development on the driver itself (or one of the other packages in this repo).

Alanzelic, 2025-06-12:
thanks ,i choose  binary packages , but the  controller_interface package is installed  , i dont know why it cant find.

adkoessler, 2025-06-12:
@urfeex, many thanks for your answer.
About the choice of building from source: I have two UR10e arms working together. From the experience I gathered on ROS1, I need to mingle with launch files so that robots do not target the same TCP ports. This explains why I need to modify the source code, but I'd gladly be proven wrong and use a simpler solution.
I could use the binary to test each robot, but I'm not sure how ROS will handle `apt delete ros-jazzy-ur` if I switch to source code for dual-arm experiments.

Continuing on the compilation problem, the update of the `hardware_interfaces` worked! However, I'm not quite there in building `ur_controllers` because a new compilation error has appeared related to `pal_statistics_msgs`:
```
Starting >>> ur_controllers
--- stderr: ur_controllers                             
In file included from /opt/ros/jazzy/include/pal_statistics/pal_statistics.hpp:38,
                 from /opt/ros/jazzy/include/pal_statistics/pal_statistics_macros.hpp:36,
                 from /opt/ros/jazzy/include/hardware_interface/hardware_interface/introspection.hpp:20,
                 from /opt/ros/jazzy/include/hardware_interface/hardware_interface/handle.hpp:31,
                 from /opt/ros/jazzy/include/controller_interface/controller_interface/controller_interface_base.hpp:25,
                 from /opt/ros/jazzy/include/controller_interface/controller_interface/controller_interface.hpp:21,
                 from /home/lcfcrobots/ur_workspace/src/Universal_Robots_ROS2_Driver/ur_controllers/src/freedrive_mode_controller.cpp:37:
/opt/ros/jazzy/include/pal_statistics/pal_statistics_utils.hpp:41:10: fatal error: pal_statistics_msgs/msg/statistics.hpp: No such file or directory
   41 | #include "pal_statistics_msgs/msg/statistics.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
I tried reproducing the approach, but much to my dismay, I have no more packages to upgrade. `pal_statistics_msgs` seems already up-to-date:
```
$ apt search jazzy-pal-statistics
Sorting... Done
Full Text Search... Done
ros-jazzy-pal-statistics/noble,now 2.6.2-1noble.20250430.001208 amd64 [installed]
  The pal_statistics package

ros-jazzy-pal-statistics-dbgsym/noble 2.6.2-1noble.20250430.001208 amd64
  debug symbols for ros-jazzy-pal-statistics

ros-jazzy-pal-statistics-msgs/noble,now 2.6.2-1noble.20250424.144219 amd64 [installed]
  Statistics msgs package

ros-jazzy-pal-statistics-msgs-dbgsym/noble 2.6.2-1noble.20250424.144219 amd64
  debug symbols for ros-jazzy-pal-statistics-msgs

$ sudo apt upgrade ros-jazzy-pal-statistics
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ros-jazzy-pal-statistics is already the newest version (2.6.2-1noble.20250430.001208).
```
I may be missing the big picture here. I'd be pleased to have your insight.

adkoessler, 2025-06-13:
Thanks @urfeex for your answer. I tried the steps you suggested, but I'm not getting there with the source code. There are so many compilation errors. I used the binaries to stop wasting our time. It builds correctly, so that's the problem solved.

I'm still not where I want to be because the robot does not move when I publish commands to the controllers. Are we still supposed to see `robot ready to receive control commands` when playing the program on the robot side? It's not showing, so I'll explore previous posts on this forum to find a solution.

About the dual-arm setup: I get your ideas, but I don't understand the code in the PR. It's sad that the person who started this couldn't finish their tutorial. Can I stick to the following way?

- create a custom package to launch both robots
- in the main launchfile, call to ur_control.launch.py with different namespaces for robot1 and robot2