UniversalRobots/Universal_Robots_ROS2_Driver #1403: Polyscope X External Control and Driver

becketps, 2025-06-14:
Ros2 Humble 
Ubuntu 22.04

Alles im Simulator für Polyscope X mit neuester Version 10.8 ohne roboter

Installierter Driver Ros2:  https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/installation/installation.html

Ich habe das Repository des ROS 2-Treibers für Universal Robots erfolgreich von folgendem Link geklont und gebaut (ROS 2 Humble):
Intallierter Driver Ros2:  https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/installation/installation.html

Versuch:
Ich habe versucht, den Driver wie folgt zu starten:
Colcon build läuft durch. 
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur20 robot_ip:=172.17.0.3  --network="host"
Antwort: 
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur20 robot_ip:=172.17.0.3  --network="host"
usage: ros2 [-h] [--use-python-default-buffering]
            Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: unrecognized arguments: --network=host

Auch führte zum Fehler
 ros2 run ur_client_library start_ursim.sh
Oder 
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur20 robot_ip:=172.17.0.3  
Quelle: https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_client_library/doc/setup/ursim_docker.html

Ergebnis:
·	Fehlerausgabe 
·	Der Browser (mit der laufenden PolyScope X Oberfläche) fror ein.
·	Es konnte keine stabile Verbindung zum Roboter aufgebaut werden.
Unklar:
·	Wann genau muss der ROS 2 Driver gestartet werden und wie?
·	Wie ist die korrekte Reihenfolge bzw. Konfiguration, wenn die Verbindung vom Host-PC zu einem simulierten Roboter (oder realen UR) erfolgen soll?

Kennen Sie eine Lösung dazu? External Controll ist installiert und die IP des Dev Contrainers Ermittelt. Im External controll Feld ist die IP des Host-PC eingetragen.

becketps, 2025-06-17:
I think that description is not suitable for the dev-Contrainer configuration. 

In the dev-container you start ./run-simulator --port 45000 

this runs one docker container and there is extra the hint that it should be started with --network="host" 

This is written here for the Polyscope X Contrainer below: 

https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCapX?tab=readme-ov-file#driver-network

how does this match with your driver?

becketps, 2025-06-17:
Dear Support,

I realize I may have misunderstood the setup process so far.

As I understand it now, you recommend running the startup script directly on the Ubuntu host machine, rather than from within the development container.

When I refer to the "dev container," I mean the one provided in the repository below, which I cloned, built, and opened in VSCode. This also opens a terminal inside the container:

      https://github.com/UniversalRobots/PolyScopeX_URCap_SDK

Could you please provide a step-by-step list of the necessary commands in this case? I’m still unsure about the correct workflow.

Additionally, I have a more general question:
In the External Control URCap, you usually have to enter your PC’s IP address and use port 50002. However, in the Docker setup, ports 30001–30004 are exposed, and I don't see where port 50002 comes into play. Could you clarify the purpose of this port and whether it’s still relevant?

Lastly, once everything is set up:

Will the topic/action list be visible as usual?

Will direct TCP/IP communication (e.g. for sending URScript commands) work the same way?

Thank you in advance for your help!

urfeex, 2025-06-18:
If you have ROS installed on your host machine, there's nothing more to do than the steps explained in https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/installation/installation.html#install-from-binary-packages and https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/usage/simulation.html#usage-with-official-ur-simulator. I've updated the simulation startup page to explicitly explain the differences with PolyScope X in #1415.

Regarding your port question:

There are two machines participated in the communication: The PC running the driver and the robot (Or URSim docker container in your case). Each of them open up ports to enable communication. The robot controller has interfaces on ports 30001-30004 that the driver connects to and the driver opens ports 50001-50004 for the robot controller (or more specifically: The script code running on the controller) to connect to. Details of that are explained in https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_client_library/doc/architecture.html.