UniversalRobots/Universal_Robots_ROS2_Driver #1416: UR3e connection overflow, robot disconnects and ROS requires full restart.

tvu-radicalai, 2025-06-17:
### Affected ROS2 Driver version(s)

3.3.0-1noble.20250520.224221

### Used ROS distribution.

Other

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with lowlatency kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.21.3 on robot

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

### Issue details

# Summary
I am running the UR3e headless over ROS 2 (jazzy, which is a slightly newer version but all functionality seems fine.

I am controlling the robot with MoveIt 2.

During operation, on my hardware (which is an Intel NUC with a modern processor) I get this issue:

```
[ros2_control_node-1] [ERROR] [1750169798.297154061] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750169798.299138637] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750169798.301144542] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
```
At this point, on the teach pendant we get the issue:

```
socket_read_binary_integer: timeout
Socket timed out waiting for connection on reverse socket. The script will exit now.
External control: stopping communication and control.
```

I am running Ubuntu 24.04 with a realtime kernel (or specifically low latency: `linux 6.12.28-rt10`, installed according to the UR client library guide).

The issue persists regardless of the state of `robot_receive_timeout` (which by default is `0.02`, and I've increased it up to the maximum of `0.2`).

## Issue details
If I echo the topic `joint_states` I get a frequency drop off as this problem occurs:
```
	min: 0.000s max: 0.006s std dev: 0.00009s window: 10000
average rate: 499.850
	min: 0.000s max: 0.006s std dev: 0.00010s window: 10000
average rate: 371.105
	min: 0.000s max: 6.943s std dev: 0.06940s window: 10000
average rate: 371.136
	min: 0.000s max: 6.943s std dev: 0.06940s window: 10000
average rate: 371.135
	min: 0.000s max: 6.943s std dev: 0.06940s window: 10000
```

## Steps to Reproduce
Launch UR3e (essentially Universal Robot's ROS driver, but we have custom code sending paths/planning collisions etc with MoveIt2, we are not using UR's MoveIt UI).

Run test script of motions (sending ROS 2 actions to set joint positions). If we run between two positions 100 times, it usually will fail (but sometimes makes it to 100)

## Expected Behavior
No failure to maintain ros2 control frequency, no disconnect of UR/reverse socket timeout.

## Actual Behavior
Robot fails execution during motion and full ROS stack needs restart to continue.

# Workaround Suggestion
No workaround found yet.

I guess we could try downgrade to ROS 22.04 but I don't think this is the issue.

### Relevant log output

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-06-17:
What does the network connection between the ROS machine and the robot look like? Is it a direct cable connection or does it go over a switch? What is the load on the ROS machine?

tvu-radicalai, 2025-06-17:
Direct cable connection over gigabit ethernet. We aren't running over a switch, and the ROS machine is running nothing else but the code for the UR3e.

tvu-radicalai, 2025-06-18:
It occurs intermittently, meaning it can occur straight away or after 100 cycles. We are not using Ubuntu server and the issue occurs when the hardware is in use and the screen is on, so I don't think power management is the culprit.

I did actually recompile the UR client library from source and attempt to increase the `robot_receive_timeout` above 0.2 to see if this helps, but it seems like this is quite an important number and the driver segfaults if I do so. (I am keen to spend a bit more time to get this to work, but it might not be possible. Even if it is, it doesn't explain why the socket times out—this seems to be the actual issue to me).

Thank you.

urfeex, 2025-06-25:
Maybe to give you some more context:

```
socket_read_binary_integer: timeout
Socket timed out waiting for connection on reverse socket. The script will exit now.
External control: stopping communication and control.
```

This occurs when the script running on the robot doesn't get a new command in time from the ROS side. The timeout for this isn't compiled into the library (that's only the default value), but it is set in the driver explicitly:

https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/98d66d07f82733885e85e1fc2ce097ad6da0d954/ur_robot_driver/src/hardware_interface.cpp#L843

That timeout can be set through a hardware parameter in the [`ros2_control.xacro`](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/98d66d07f82733885e85e1fc2ce097ad6da0d954/ur_robot_driver/urdf/ur.ros2_control.xacro#L28). Changing the default in the client library shouldn't make your driver crash.

On the other hand,

```
[ros2_control_node-1] [ERROR] [1750169798.301144542] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
```

indicates that the control loop running on the ROS pc is not running reliably with 500 Hz. The robot is sending data to the ROS machine with 500 Hz where data is stored into a queue. The control loop pops items from that queue. If items are not removed from the queue fast enough, the queue will get filled up until it eventually has no more capacity which is when this error is triggered.

The reasons why your control loop is not keeping up 500 Hz can be for example:

- Not enough hardware resources available (seems not to be the case for you?)
- Scheduling / thread priority issues -> Did you enable FIFO scheduling with high priority as explained in https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_client_library/doc/real_time.html#setup-user-privileges-to-use-real-time-scheduling?
- Some operation in the cycle takes too long
  - Any other hardware component's `read()` or `write()` operation takes time. Do you have other hardware with ros2_control in your system?
  - Any controller's `update()` takes too long

Both issues are symptoms from a control loop not running reliably at 500 Hz, so that might be the top priority to debug.

urfeex, 2025-06-26:
That high stddev on the JTC is definitively worrying and could be the source of problems. Those stats unfortunately don't show how often a higher execution time occurs, but that execution time of 291 ms would definitively cause problems, as this is equivalent to ~ 150 control cycles. Given the average of 16.6 us it was probably indeed a single outlier.

How large are the trajectories (number of points) that you send to the controller? I remember an issue about long trajectories in the JTC (https://github.com/ros-controls/ros2_controllers/issues/1293) that showed a high spike in the cycle time once a new trajectory was received (though it was "only" 8 instead of 4 ms there).

As a workaround you could try to use the passthrough_trajectory_controller instead of the scaled jtc. It acts as a drop-in replacement and loads trajectory execution to the robot controller rather than the ROS machine. The numbers you posted seem to indicate that the problem could be somewhere in the JTC, so I would be curious whether that changes something.

urfeex, 2025-07-08:
I'm happy that the passthrough trajectory controller is a working workaround for you, but I agree that the initial topic isn't resolved and should get investigated further. I'm still in the process of setting up a long term test in order to be able to run those long-running tests without blocking my dev machine.

> when running we get trajectories selecting 'quintic splines' I have observed others running with cubic—is this configurable in some way?

When providing setpoints with positions and velocities, cubic splines will be used. When also providing accelerations, quintic splines will be used. The latter should be smoother, so they should be preferable if your have accelerations in your setpoints available.