UniversalRobots/Universal_Robots_ROS2_Driver #1423: Issue name

becketps, 2025-06-22:
### Affected ROS2 Driver version(s)

humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

URSim in linux

### Robot SW / URSim version(s)

10.8

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I just cloned the repository branch for ros2 humble and  when doing colcon build there is an error.

## Issue details
 source /opt/ros/humble/setup.bash
colcon build
Starting >>> ur_dashboard_msgs
Starting >>> ur_moveit_config
Finished <<< ur_moveit_config [0.08s]
Finished <<< ur_dashboard_msgs [0.32s]                     
Starting >>> ur_controllers
--- stderr: ur_controllers                         
CMake Error at CMakeLists.txt:95 (add_library):
  Target "ur_controllers" links to target
  "controller_interface::controller_interface" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "test_load_force_mode_controller" links to target
  "controller_interface::controller_interface" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:178 (ament_add_gmock)
## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
build successful does never appear

### Relevant log output

```shell
source /opt/ros/humble/setup.bash
colcon build
Starting >>> ur_dashboard_msgs
Starting >>> ur_moveit_config
Finished <<< ur_moveit_config [0.08s]
Finished <<< ur_dashboard_msgs [0.32s]                     
Starting >>> ur_controllers
--- stderr: ur_controllers                         
CMake Error at CMakeLists.txt:95 (add_library):
  Target "ur_controllers" links to target
  "controller_interface::controller_interface" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "test_load_force_mode_controller" links to target
  "controller_interface::controller_interface" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:178 (ament_add_gmock)

CMake Error at /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "test_load_force_mode_controller" links to target
  "controller_manager::controller_manager" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:178 (ament_add_gmock)

CMake Error at /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "test_load_force_mode_controller" links to target
  "ros2_control_test_assets::ros2_control_test_assets" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:178 (ament_add_gmock)

CMake Error at /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "test_load_freedrive_mode_controller" links to target
  "controller_interface::controller_interface" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:186 (ament_add_gmock)

CMake Error at /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "test_load_freedrive_mode_controller" links to target
  "controller_manager::controller_manager" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:186 (ament_add_gmock)

CMake Error at /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "test_load_freedrive_mode_controller" links to target
  "ros2_control_test_assets::ros2_control_test_assets" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:186 (ament_add_gmock)

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< ur_controllers [0.51s, exited with code 1]

Summary: 2 packages finished [1.02s]
  1 package failed: ur_controllers
  1 package had stderr output: ur_controllers
  3 packages not processed
```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-06-25:
Are you building from the `humble` branch? Is your system up to date?

UDAYAMANOJ, 2025-06-27:
Build failed in **Jazzy** as well. 

Any solutions?

Starting >>> ur_dashboard_msgs
Starting >>> ur_moveit_config
Finished <<< ur_moveit_config [1.93s]                                                               
[Processing: ur_dashboard_msgs]                             
[Processing: ur_dashboard_msgs]                                     
Finished <<< ur_dashboard_msgs [1min 7s]                             
Starting >>> ur_controllers
[Processing: ur_controllers]                                  
[Processing: ur_controllers]                                     
[Processing: ur_controllers]                                       
--- stderr: ur_controllers                                         
/home/manoj/ros2_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/manoj/ros2_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:118:76: error: no match for ‘operator&&’ (operand types are ‘realtime_tools::RealtimeBuffer<bool>’ and ‘bool’)
  118 |   if (current_trajectory_msg != *new_external_msg && (rt_has_pending_goal_ && !active_goal) == false) {
      |                                                       ~~~~~~~~~~~~~~~~~~~~ ^~ ~~~~~~~~~~~~
      |                                                       |                       |
      |                                                       |                       bool
      |                                                       realtime_tools::RealtimeBuffer<bool>
/home/manoj/ros2_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:118:76: note: candidate: ‘operator&&(bool, bool)’ (built-in)
  118 |   if (current_trajectory_msg != *new_external_msg && (rt_has_pending_goal_ && !active_goal) == false) {
      |                                                       ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/home/manoj/ros2_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:118:76: note:   no known conversion for argument 1 from ‘realtime_tools::RealtimeBuffer<bool>’ to ‘bool’
/home/manoj/ros2_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:178:33: error: no match for ‘operator!’ (operand type is ‘realtime_tools::RealtimeBuffer<bool>’)
  178 |       if (!before_last_point && !rt_is_holding_ && cmd_timeout_ > 0.0 && time_difference > cmd_timeout_) {
      |                                 ^~~~~~~~~~~~~~~
/home/manoj/ros2_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:178:33: note: candidate: ‘operator!(bool)’ (built-in)
/home/manoj/ros2_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:178:33: note:   no known conversion for argument 1 from ‘realtime_tools::RealtimeBuffer<bool>’ to ‘bool’
In file included from /opt/ros/jazzy/include/libstatistics_collector/libstatistics_collector/moving_average_statistics/moving_average.hpp:30,
                 from /opt/ros/jazzy/include/libstatistics_collector/libstatistics_collector/collector/collector.hpp:22,
                 from /opt/ros/jazzy/include/libstatistics_collector/libstatistics_collector/topic_statistics_collector/topic_statistics_collector.hpp:26,
                 from /opt/ros/jazzy/include/libstatistics_collector/libstatistics_collector/topic_statistics_collector/received_message_age.hpp:26,
                 from /opt/ros/jazzy/include/rclcpp/rclcpp/topic_statistics/subscription_topic_statistics.hpp:26,
                 from /opt/ros/jazzy/include/rclcpp/rclcpp/subscription.hpp:50,
                 from /opt/ros/jazzy/include/rclcpp/rclcpp/any_executable.hpp:25,
                 from /opt/ros/jazzy/include/rclcpp/rclcpp/memory_strategy.hpp:25,
                 from /opt/ros/jazzy/include/rclcpp/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/jazzy/include/rclcpp/rclcpp/executor_options.hpp:22,
                 from /opt/ros/jazzy/include/rclcpp/rclcpp/executor.hpp:38,
                 from /opt/ros/jazzy/include/rclcpp/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/jazzy/include/rclcpp/rclcpp/executors.hpp:21,
                 from /opt/ros/jazzy/include/rclcpp/rclcpp/rclcpp.hpp:172,
                 from /opt/ros/jazzy/include/pal_statistics/pal_statistics_utils.hpp:45,
                 from /opt/ros/jazzy/include/pal_statistics/pal_statistics.hpp:38,
                 from /opt/ros/jazzy/include/pal_statistics/pal_statistics_macros.hpp:36,
                 from /opt/ros/jazzy/include/hardware_interface/hardware_interface/introspection.hpp:20,
                 from /opt/ros/jazzy/include/hardware_interface/hardware_interface/handle.hpp:31,
                 from /opt/ros/jazzy/include/controller_interface/controller_interface/controller_interface_base.hpp:25,
                 from /opt/ros/jazzy/include/controller_interface/controller_interface/controller_interface.hpp:21,
                 from /opt/ros/jazzy/include/joint_trajectory_controller/joint_trajectory_controller/joint_trajectory_controller.hpp:27,
                 from /home/manoj/ros2_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/scaled_joint_trajectory_controller.hpp:43,
[... cut ...]
  370 |       pids_[i] = std::make_shared<control_toolbox::Pid>(gains.p, gains.i, gains.d, gains.u_clamp_max, gains.u_clamp_min,
      |                                                                                                             ^~~~~~~~~~~
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:104: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:409: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ur_controllers [1min 40s, exited with code 2]

becketps, 2025-06-28:
implementation as described still does not work as expected.

What I Did:
  1. Installed from Binary (ROS2 Rolling)
  As mentioned, I only followed the "Install from binary packages" instructions:
  
    sudo apt-get install ros-${ROS_DISTRO}-ur
  
The driver installation completed without issues.
  
  2. Docker Environment SetupFully reinstalled Docker and Docker Desktop
           
       Verified the firewall is disabled:

      sudo ufw status
      Status: inactive

Confirmed that Docker itself is not the problem: even on a freshly installed system, I encounter the same issue:

3. Start your skript loads the docker contrainer but does not end with setup: 

https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/scripts/start_ursim.sh

         ./start_ursim.sh -v 10.8.0 -m ur20
        Output:
        
        ROBOT_MODEL: ur20
        ROBOT_SERIES: polyscopex
        URSIM_VERSION: 10.8.0
        ...
        Starting URSim. Waiting for UrService to be up...........

→ It never gets past this point.

🧪 Alternative Test (Docker Image via Docker Hub)
Tried the universalrobots/ursim_polyscopex Docker image, similar to the DevContainer setup.

Source: https://hub.docker.com/r/universalrobots/ursim_polyscopex

Command used:

docker run --rm -it -p 80:80 --add-host "host.docker.internal:host-gateway" --env HOST_ARCH=amd64 --network bridge --privileged universalrobots/ursim_polyscopex:0.14.47

This does successfully launch the simulator:

The Polyscope X UI is visible under http://localhost:80

Installed the External Control URCap

Retrieved host IP using ifconfig and added it in the URCap

Identified container IP via:

docker inspect -f '{{range .NetworkSettings.Networks}}{{.IPAddress}}{{end}}' <container_name_or_id>

Launched the driver with:

ros2 launch ur_robot_driver ur_control.launch.py \
    ur_type:=ur20 \
    robot_ip:=<CONTAINER_IP> \
    launch_rviz:=true
Set robot state to Active in the browser UI

Under “Program” → selected External Control → clicked “Play”

❌ Result: A message appears saying connection could not be established.

🔄 Conclusion
The Docker approach behaves very similarly to the DevContainer with the SDK, leading to identical issues in connectivity.

urfeex, 2025-07-01:
When you are using the host network for starting up the simulator, a lot of things change. You will probably have to use the localhost ip (127.0.0.1) everywhere. When starting up the driver, when configuring the URCap, potentially also pass it explicitly as reverse IP. That's why I would prefer sticking to the suggested workflow, as this way we can focus on getting your Docker setup running correctly rather than opening topics on the side everywhere. In the end, you want a solution, where the simulator is exchangable with a real robot very easily, right?

So, let's focus on the fact that you cannot reach your docker container. 

Try starting a simple nginx server in the ursim_net:

```
docker run -it --rm --net ursim_net --name web nginx
```

With that started in one shell, open another shell. There run

```
docker container inspect web | grep IPAddres
```

The result probably looks something like this:

```
            "SecondaryIPAddresses": null,
            "IPAddress": "",
                    "IPAddress": "192.168.56.2",
```

Now, that we have the IP address of the container in our ursim_net, let's try to reach it. Let's try to ping it and using curl to access the http web page:

```console
$ ping 192.168.56.2
PING 192.168.56.2 (192.168.56.2) 56(84) bytes of data.
64 bytes from 192.168.56.2: icmp_seq=1 ttl=64 time=0.080 ms
64 bytes from 192.168.56.2: icmp_seq=2 ttl=64 time=0.085 ms
^C
--- 192.168.56.2 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1023ms
rtt min/avg/max/mdev = 0.080/0.082/0.085/0.002 ms
```

```console
$ curl 192.168.56.2
<!DOCTYPE html>
<html>
<head>
<title>Welcome to nginx!</title>
<style>
html { color-scheme: light dark; }
body { width: 35em; margin: 0 auto;
font-family: Tahoma, Verdana, Arial, sans-serif; }
</style>
</head>
<body>
<h1>Welcome to nginx!</h1>
<p>If you see this page, the nginx web server is successfully installed and
working. Further configuration is required.</p>

<p>For online documentation and support please refer to
<a href="http://nginx.org/">nginx.org</a>.<br/>
Commercial support is available at
<a href="http://nginx.com/">nginx.com</a>.</p>

<p><em>Thank you for using nginx.</em></p>
</body>
</html>
```

If those two commands don't work, stop here. There's something broken with your docker setup that we need to fix first. You can make a quick test recreating your ursim_net (stop your nginx container first)
```
docker network remove ursim_net
docker network create --subnet=192.168.56.0/24 ursim_net
```

Then, try the same thing again. If it doesn't work, please post the output of `docker network inspect ursim_net`. [This post](https://forums.unraid.net/topic/135205-unraid-cant-ping-docker-container-using-network-ip/) suggests, there is a Docker setting for that. I have never used Docker Desktop myself, so I don't know anything about this. If you do use Docker Desktop, may I suggest trying to use Docker engine? As far as I know, you should be able to switch between the two as explained in the installation: https://docs.docker.com/desktop/setup/install/linux/

becketps, 2025-07-02:
Yes I started the ursim and configured everything with the ip. Then I did the ros2 command and then I pressed the play botton. 

The same thing I did with a normal TCP IP Programm in Python which runs on Polyscope 5 as well but did not succed to be delivered on polyscope X

urfeex, 2025-07-03:
> Yes I started the ursim and configured everything with the ip. Then I did the ros2 command and then I pressed the play botton.

Let's stick to that, please. You did

1. start ursim with the start script from the client library
2. Create a program with the external control node, saying it's going to connect to `192.168.56.1:50002`
3. start the ros2 driver using `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.56.101`
4. Press the "Update Program" button on the teach pendant next to the program node

With that, please answer the following questions:
- Does the driver's shell print 
  ```
  [ros2_control_node-1] [INFO] [1751534527.917598277] [UR_Client_Library:]: Robot requested program
  [ros2_control_node-1] [INFO] [1751534527.917794296] [UR_Client_Library:]: Sent program to robot
  ```
  If it does, everything should potentially work. When you press "play", the driver should print 
  ```
  [ros2_control_node-1] [INFO] [1751534636.006663534] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.
  ```
- If the that doesn't work, please open http://192.168.56.101/universal-robots/external-control/external-control-backend/rest-api/50002/192.168.56.1/ with a web browser and post the response here.

becketps, 2025-07-04:
Thanks. It did work. Why is it the IP Adress in the external Control of 192.168.56.1:50002?
Is there any documentation how to figure out what all the services and Actions are doing and which parameters they expect? Do you have a suggestion how to get used to it. 

I also started my pyhton Script and on Port 30002 ist succeds. I'm not sure why only this port is usable not 30004. I don't know wy I'm wondering that I should not use the [ScriptSender](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_client_library/doc/architecture/script_sender.html) and it worked. 

Why isn't it possible with the [polyscope X sdk](https://github.com/UniversalRobots/PolyScopeX_URCap_SDK) of UR. I think the sample-ros2 is a bit the same.

urfeex, 2025-07-04:
> Why is it the IP Adress in the external Control of 192.168.56.1:50002? and not the ip which I see in ipconfig? From my understanding it should be the host right?

Yes, exactly! It should be the host, given an IP address that the robot (in this case the Docker container) can reach. Since the docker container is put on a `192.168.56.0/24` network with the host being `192.168.56.1`, that one has to be used.

> Is there any documentation how to figure out what all the services and Actions are doing and which parameters they expect? Do you have a suggestion how to get used to it and where to find them.

Please have a look in our documentation: The [usage -> controllers](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/usage/controllers.html) page lists all controllers, where for each controller there's their respective documentation linked e.g. the [force_mode_controller(https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_controllers/doc/index.html#force-mode-controller)

> I also started my python Script and on Port 30002 ist succeds. I'm not sure why only this port is usable not 30004.

Because port 30004 is for the RTDE interface, while 30002 is the secondary interface, see [here](https://www.universal-robots.com/articles/ur/interface-communication/overview-of-client-interfaces/)

> Why isn't it possible with the [polyscope X sdk](https://github.com/UniversalRobots/PolyScopeX_URCap_SDK) of UR. I assume the sample-ros2 is a bit the same.

I don't quite understand what you mean with that. It is definitively possible to run the simulator from the SDK, but then you would have to fiddle around with port and IP setup quite a lot more.

Since things seem to be working now, I will go ahead and close this issue.

closed 2025-07-04 by urfeex