UniversalRobots/Universal_Robots_ROS2_Driver #1457: Issue name

karanchahal-nv, 2025-07-14:
### Affected ROS2 Driver version(s)

jazzy

### Used ROS distribution.

Other

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

Polyscope 5.22

### How is the ROS driver used.

Headless without using the teach pendant, Through the robot teach pendant using External Control URCap

### Issue details

# Summary
We are trying to launch the UR drivers on Polyscope 5.22 but using ROS2 Jazzy instead of ROS2 Humble.

We realized that the `ur_ros2_control_node` node inside the `ur_robot_driver` package does not exist anymore.

Could you help us in pointing out why the change was made and what we should do to properly launch the ur driver now ?

Our earlier code looked like this:

```
ur_control_node = Node(
   package='ur_robot_driver',
   executable='ur_ros2_control_node',
   parameters=[
      robot_description,
      update_rate_config_file,
      ParameterFile(initial_joint_controllers, allow_substs=True),
      ParameterFile(robotiq_gripper_controllers, allow_substs=True),
   ],
   output='screen',
   on_exit=Shutdown(),
)
```
## Issue details
*Detailed description help us understand the problem. Code are welcome!*

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
*What did you expect and why?*

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

# Workaround Suggestion
*If a workaround has been found, you are welcome to share it.*

### Relevant log output

```shell
[tool_communication.py-1] [INFO] [1752017837.261150633] [tool_communication]: 192.56.1.2
[tool_communication.py-1] [INFO] [1752017837.262513545] [tool_communication]: Remote device is available at '/tmp/ttyUR'
[tool_communication.py-1] [INFO] [1752017837.263205417] [tool_communication]: Starting socat with following command:
[tool_communication.py-1] socat pty,link=/tmp/ttyUR,raw,ignoreeof,waitslave tcp:192.56.1.2:54321
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'ur_ros2_control_node' not found on the libexec directory '/opt/ros/jazzy/lib/ur_robot_driver'
```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-07-15:
Hi,

first of all sorry for not further documenting that change. It was introduced in #939 and we didn't track the migration log back then. I'll make a PR adding it to the migration notes.

The short answer is to do it like it is done in https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/2303a43d7490d0b1f7c482fcc5f0ff3e975ce816/ur_robot_driver/launch/ur_control.launch.py#L70-L80

The `ros2_control_node` from the `controller_manager` package supports all features we needed, so we switched back to using the upstream one.

All necessary information is passed through the xacro ros2_control tag (which also moved in the migration to Jazzy, see [the migration notes](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/migration/jazzy.html#ros2-control-xacro-tag-moved-to-driver-package)).

urfeex, 2025-07-15:
@karanchahal-nv I've added a section in the migration notes in #1458. Would that help or do you have any comments on that?

karanchahal-nv, 2025-07-15:
great ! thank you ! I will try this out and revert back with the results.

karanchahal-nv, 2025-07-17:
Gist Link of launch file (with the gripper controllers): https://gist.github.com/karanchahal-nv/00f0d893ecbf2bbbcae871b43dea1f22

urfeex, 2025-07-17:
I think, the URDF (respectively your ros2_control definition) is the interesting part. Why is gpio/standard_digital_output_cmd_0 not there?

It is defined here: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/32cb2534b0cb4263b2c97fbdc3039d4925c0a244/ur_robot_driver/urdf/ur.ros2_control.xacro#L96

I assume, you generated your own description with your own ros2_control definition?

karanchahal-nv, 2025-07-23:
Hi, so I have some progress here, the one issue was that it seems like `ur.urdf.xacro` is moved from `ur_description` to `ur_robot_driver` - which was a little tricky to find out.

Now I am at the phase where I can load the drivers fine in our custom setup along with the robotiq gripper. Now I am trying to make the robot move.

The controller seems to be emitting this warning log:

```
[ros2_control_node-9] [WARN] [1753286859.799921021] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 500 Hz. The loop took 2.094208 ms (missed cycles : 2).
```

I am not sure why this is happening. Could this be that the `external_control` ur_script is different between jazzy and humble ?

I am using [this launch file](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/jazzy/ur_moveit_config/launch/ur_moveit.launch.py) to launch the ur move group and [this](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/jazzy/ur_robot_driver/launch/ur_control.launch.py) file to launch the drivers.

Also, maybe a bigger issue--> the scaled joint trajectory controller says that it can't accept a trajectory since the last joint velocity in trajectory is non-zero.

This used to work in Humble (we haven't switched our planner)- is this a new requirement to `scaled_joint_trajectory_controller` ?

urfeex, 2025-07-24:
> Hi, so I have some progress here, the one issue was that it seems like `ur.urdf.xacro` is moved from `ur_description` to `ur_robot_driver` - which was a little tricky to find out.

That's why I mentioned that and linked the migration notes in https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/1457#issuecomment-3072175072 ;-)

> The controller seems to be emitting this warning log:
>
> ```
> [ros2_control_node-9] [WARN] [1753286859.799921021] [controller_manager]: Overrun detected! The controller manager missed its desired rate [...]

That's something that always happened. It's only the warning output that is new. As long as it stays in that range it shouldn't be much of an issue unless it is happening constantly. The `external_control.urscript` is exactly the same for all ROS versions, as that's part of the client library which is the same version for all ROS versions.

Notes that every component (UR ROS Driver, gripper driver, any controller running) contribute to the cycle time. In the issue description you wrote that you use a standard kernel? We highly suggest using a lowlatency or preempt-rt kernel and setup the correct user rights to do FIFO scheduling with high thread priorities for the main important threads.

> Also, maybe a bigger issue--> the scaled joint trajectory controller says that it can't accept a trajectory since the last joint velocity in trajectory is non-zero.
> 
> This used to work in Humble (we haven't switched our planner)- is this a new requirement to `scaled_joint_trajectory_controller` ?

That was a change happening upstream in the `joint_trajectory_controller` in https://github.com/ros-controls/ros2_controllers/pull/834. You can change the behavior back by setting the `allow_nonzero_velocity_at_trajectory_end` to `true`. See https://control.ros.org/jazzy/doc/ros2_controllers/joint_trajectory_controller/doc/parameters.html

karanchahal-nv, 2025-07-24:
Awesome, that clears all my questions. I am able to run the drivers correctly on Jazzy now, thank you so much !!

closed 2025-07-24 by karanchahal-nv