UniversalRobots/Universal_Robots_ROS2_Driver #1460: Joint Tracking controller not reacting to /joint_tracking_controller/joint_trajectory message

GiantMGG, 2025-07-18:
### Affected ROS2 Driver version(s)

2.8.1-1 jammy

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Docker

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot, URSim in docker

### Robot SW / URSim version(s)

5.21

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I am using the joint trajectory controller for a UR5e. I am planning joint trajectories in a MPC (model predictive control) manner with 20 Hz which I send over the /joint_trajectory_controller/joint_trajectory controller. Most of the time this works well. But sometimes the joint_trajectory_controller does not react to the messages I send, but there is no error message. A pure restart solves the issue. I cannot reliably reproduce this issue.

## Issue details
I am using this command "ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=xxx.xxx.xxx.xxx initial_joint_controller:=joint_trajectory_controller use_fake_hardware:=true launch_rviz:=true" to start the mockup robot. The same issue also persists when using the real robot, though. 

On the topic /joint_trajectory_controller/controller_state, the desired trajectories are not displayed. In the case, when the controller reacts to my trajectories, I can see my trajectories under "desired". So listening to this topic is the only way I found so far to narrow down the problem to the joint_trajectory_controller.
With "ros2 control list_controllers", the joint_trajectory_controller is listed as active.

I am running everything inside a docker container with Ubuntu 22. My setup used to work reliably until I made some changes recently, including upgrading from an old ROS2 UR_driver version. 

I have also tried to send just a single trajectory to the controller via cli, but that did not get the controller to accept the trajectory. 

## Steps to Reproduce
I cannot reliably reproduce this issue. Most of the time rebuilding the docker container solves the issue. Most of the time just restarting the ROS node for the UR_driver solves the issue as well.

## Expected Behavior
I would expect the joint trajectory controller to accept and execute the trajectories it recieves, or at least print error messages.

## Actual Behavior
I attached 2 log outputs: One in the case where the controller reacted to commanded trajectories, and one, where it did not react.

# Workaround Suggestion
Restarting or rebuilding the docker container

### Relevant log output

```shell
In this case, the controller did not react to the commanded trajectories:

ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=xxx.xxx.xxx.xxx initial_joint_controller:=joint_trajectory_controller use_fake_hardware:=true launch_rviz:=true
groot@ess-kerbin:~/telemanipulation_ws$ ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=xxx.xxx.xxx.xxx initial_joint_controller:=joint_trajectory_controller use_fake_hardware:=true launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/groot/.ros/log/2025-07-18-11-44-16-390348-ess-kerbin-26048
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [26237]
[INFO] [urscript_interface-2]: process started with pid [26239]
[INFO] [robot_state_publisher-3]: process started with pid [26241]
[INFO] [rviz2-4]: process started with pid [26243]
[INFO] [spawner-5]: process started with pid [26245]
[INFO] [spawner-6]: process started with pid [26247]
[... 187 lines cut ...]
[ros2_control_node-1] [INFO] [1752839141.597943264] [controller_manager]: Loading controller 'tool_contact_controller'
[spawner-5] [INFO] [1752839141.598444403] [spawner_joint_state_broadcaster]: Configured and activated force_torque_sensor_broadcaster
[spawner-6] [INFO] [1752839141.604504370] [spawner_scaled_joint_trajectory_controller]: Loaded tool_contact_controller
[ros2_control_node-1] [INFO] [1752839141.604713910] [controller_manager]: Configuring controller 'tool_contact_controller'
[ros2_control_node-1] [INFO] [1752839141.609889489] [controller_manager]: Loading controller 'ur_configuration_controller'
[spawner-5] [INFO] [1752839141.618475200] [spawner_joint_state_broadcaster]: Loaded ur_configuration_controller
[ros2_control_node-1] [INFO] [1752839141.618781010] [controller_manager]: Configuring controller 'ur_configuration_controller'
[spawner-5] [INFO] [1752839141.626358774] [spawner_joint_state_broadcaster]: Configured and activated ur_configuration_controller
[ros2_control_node-1] [INFO] [1752839141.627450041] [controller_manager]: Loading controller 'joint_trajectory_controller'
[ros2_control_node-1] [WARN] [1752839141.636709482] [joint_trajectory_controller]: [Deprecated]: "allow_nonzero_velocity_at_trajectory_end" is set to true. The default behavior will change to false.
[spawner-5] [INFO] [1752839141.638624108] [spawner_joint_state_broadcaster]: Loaded joint_trajectory_controller
[ros2_control_node-1] [INFO] [1752839141.638995857] [controller_manager]: Configuring controller 'joint_trajectory_controller'
[ros2_control_node-1] [INFO] [1752839141.639100357] [joint_trajectory_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ros2_control_node-1] [INFO] [1752839141.639111077] [joint_trajectory_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[ros2_control_node-1] [INFO] [1752839141.639121167] [joint_trajectory_controller]: Using 'splines' interpolation method.
[ros2_control_node-1] [INFO] [1752839141.639770375] [joint_trajectory_controller]: Controller state will be published at 100.00 Hz.
[ros2_control_node-1] [INFO] [1752839141.641866021] [joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[spawner-5] [INFO] [1752839141.652789508] [spawner_joint_state_broadcaster]: Configured and activated joint_trajectory_controller
[INFO] [spawner-6]: process has finished cleanly [pid 27646]
[INFO] [spawner-5]: process has finished cleanly [pid 27644]
```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-07-18:
Just for clarification: You use the command topic, not the action interface, right? It appears that there are issues with the controller's command topic, as reported in #1459 and #1446. Is this potentially a duplicate of one of these?

Currently, to me this looks like a regression in the controller, not necessarily the driver. Especially since you seem to be able to reproduce that using mock hardware.

urfeex, 2025-07-18:
Update: This has already been reported and fixed in https://github.com/ros-controls/ros2_controllers/pull/1749 (or https://github.com/ros-controls/ros2_controllers/pull/1765 for Humble). There is a [new sync coming up today](https://discourse.openrobotics.org/t/preparing-for-humble-sync-and-patch-release-2025-07-18/45250) containing this fix. So, you should be able to update tomorrow to get that problem fixed.

GiantMGG, 2025-07-21:
Thanks for the support. The problem seems to be fixed now.

The sync is (to my understanding) still pending. So I installed the latest JTC from source (https://github.com/ros-controls/ros2_controllers/tree/f1bc4f8cd7d3ba9f5e2c79ca2515f28c32c4691d).

This issue can be closed.

closed 2025-07-21 by urfeex