UniversalRobots/Universal_Robots_ROS2_Driver #1463: Simulated Robot - operation mode password

adammetcalf, 2025-07-22:
I am using ROS2 Humble on Ubuntu 22.

I am using the simulated robot following the instructions on this page here: https://docs.ros.org/en/rolling/p/ur_robot_driver/doc/usage/simulation.html

I have installed Docker, and started the Docker container:

```
$ ros2 run ur_client_library start_ursim.sh -v 10.8.0 -m ur3e
ROBOT_MODEL: ur3e
ROBOT_SERIES: polyscopex
URSIM_VERSION: 10.8.0
330bac62ba6795da00e4f465b156dd9511c298e30557360a03890b9ffa528992
Starting URSim. Waiting for UrService to be up..................................
UrService is up
Installing URCapX /home/adam/.ursim/polyscopex/urcaps/external-control-1.0.0.urcapx

To access PolyScopeX, open the following URL in a web browser.

	http://192.168.56.101

To exit, press CTRL+C
```

I have connected to the local web interface. Everything has launched correctly.

When I attempt to change the operational mode to 'Automatic', a password is required. The default password is not accepted.

[image]

urfeex, 2025-07-22:
The default password there is `operator`. Note, however, you will not have to change to automatic mode / remote control mode in the simulator.

adammetcalf, 2025-07-22:
Fantastic, thanks very much! I really appreciate the super quick response.

I may well not need to access other operational modes in simulation, but its certainly removed a potential point of failure in the work I'm currently doing. Thanks again!