UniversalRobots/Universal_Robots_ROS2_Driver #1513: UR3e wrist_3_joint infinite joint never works properly

rjbaw, 2025-09-26:
### Affected ROS2 Driver version(s)

1noble.20250624

### Used ROS distribution.

Jazzy

### Which combination of platform is the ROS driver running on.

Docker

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

URSim in docker

### Robot SW / URSim version(s)

5.23

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

# Summary
UR3 and UR3e according to the specifications has infinite end effector wrist but it never seems to work with ROS2 due to a combination of issues. 

## Issue details
It is well known that robot planners struggle at +-180 degrees wrap-around. This causes UR3e to not perform according to specifications. If the robot does execute on these bad trajectories, it causes the robot to be in a irrecoverable state (INVALID_START_STATE). Only when restarting the Robot/URSim completely can I start planning/executing trajectories again.

### Planning using different planners

https://github.com/user-attachments/assets/51585c33-efb6-4d29-b965-5de4224710e7

### Incrementing joints directly

https://github.com/user-attachments/assets/12b46476-8674-4c75-a242-8d1e74bf173b

### Invalid start state

https://github.com/user-attachments/assets/a8a5f452-f1d3-4099-9905-a30b1887aef9

## Steps to Reproduce

1. Start the UR3-e URSim
2. Install `ros-jazzy-ur` package
3. Launch the UR driver using `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.56.101 headless_mode:=true`
4. Launch the MoveIt driver using `ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e launch_rviz:=true`

## Expected Behavior
Infinite joint should work or should be documented that it does not work with ROS2 or specific planners.

# Workaround Suggestion
I am able to replicate UR3e infinite wrist in URScript but it is less than ideal.

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

rjbaw, 2025-10-01:
Thanks for the reply. That makes the issue very clear. 
I have already been avoiding bad trajectories with path constraints but the wrap-around issue has been persistent issue, which requires a return to start state operation on every robot run. 
I have tested your workaround and it does increase the operating joint space to -2pi to 2pi, so that is very helpful.

urfeex, 2025-10-06:
One more side note: In the workaround I suggested using min/max positions of 360deg. From a URDF standpoint, any limit would probably be working fine, so you could also set it to 36000 to gain 100 full revelations in each direction. I am, however, not sure what planners will do with that, as a Cartesian pose will now have a lot of possible solutions.