UniversalRobots/Universal_Robots_ROS2_Driver #1514: Ros2 Driver on Jetson Orin AGX

becketps, 2025-09-28:
### Affected ROS2 Driver version(s)

Humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Other Linux system

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

URSim in linux

### Robot SW / URSim version(s)

Polyscope X 10.10

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
*Short introduction to the issue and how it impact you and why*
I wanted to build the Ros2 Humble Driver on the Jetson AGX running with Jetpack 6.4.4

## Issue details
*Detailed description help us understand the problem. Code are welcome!*

The UR Libary is installed properly but the ros2 Folder gives me such errors 

[build_log.txt](https://github.com/user-attachments/files/22586065/build_log.txt)

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
*What did you expect and why?*

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

# Workaround Suggestion
*If a workaround has been found, you are welcome to share it.*

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-09-30:
> ```
>  In file included from /home/jetson/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/ur_configuration_controller.cpp:41:
>  /home/jetson/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/ur_configuration_controller.hpp:47:10:   > fatal error: realtime_tools/realtime_thread_safe_box.hpp: No such file or directory
>   47 | #include <realtime_tools/realtime_thread_safe_box.hpp>
```

That looks like you are trying to compile the main branch. If you are on Humble, please use the humble branch.

We highly suggest to use the binary packages instead of compiling things from source.

becketps, 2025-10-18:
When is the ursim Simulator for polyscope X running on Jetson AGX?